FreeCAD/FreeCAD #16776: App crashes with odd pop-up dialog handling after variable set input validation error

shevron, 2024-09-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I am running FreeCAD on Mac (M1), and just decided to try 1.0.0RC1. I am not sure if this is a general problem with all input validation error messages, but I can consistently crash the app when doing the following:

1. Start FreeCAD and create an empty document
2. Click on the `{}` button - "Create a Variable Set" 
3. Don't type in a Name
4. Click on the "Group" dropdown menu (click on the triangle)
5. **Issue 1**: I get two pop-up dialogues showing the same error message (see attached screenshot)
6. I can't click on anything to close the dialogues. They seem to "lock eachother" out
7. Hitting Spacebar on the keyboard closes the dialogue, and then the app immediately crashes
 
I'm also attaching the Apple crash report, if it helps: [freecad-1.0.0rc1-crash-report.txt](https://github.com/user-attachments/files/17112957/freecad-1.0.0rc1-crash-report.txt)

[image]

### Full version info

```shell
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.2.0
  * fasteners 0.5.25
  * CurvedShapes 1.0.9
  * Assembly4 0.50.15
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-24:
@johnocfii do you mind reproducing on macOS ? TIA!

luzpaz, 2024-09-24:
https://github.com/FreeCAD/FreeCAD/blob/1f4adf9e56d19fee25effd0a5553a757e4113438/src/Gui/DlgAddPropertyVarSet.cpp#L348-L349 

An aside, the newline in the error message should be removed, IMHO

Roy-043, 2024-09-24:
Related: #16189.

JohnOCFII, 2024-09-24:
> @JohnOCFII do you mind reproducing on macOS ? TIA!

I've not used VarSets yet (I'm a Spreadsheet man), and don't see the `{}` button.  Is this related to a specific workbench?  Do I need to have already created some objects?
[image]

shevron, 2024-09-24:
@JohnOCFII for me it is displayed by default in the Part Design workbench, but maybe this is some old setting I had. I don't use VarSets either, I just happened to click it. 

The button is a part of the "Structure" toolbar. 
[image]

JohnOCFII, 2024-09-24:
> The button is a part of the "Structure" toolbar.

I wasn't displaying the `Structure` toolbar.  (Not sure if that was a conscious choice, or a theme setting, or something else I just turned off since I didn't think I needed those options).

JohnOCFII, 2024-09-24:
I can confirm the issue.

Crash Report:

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               freecad [16620]
Path:                  /Applications/Other Apps/FreeCAD-38827.app/Contents/Resources/bin/freecad
Identifier:            freecad
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-09-24 16:30:16.6903 -0500
OS Version:            macOS 15.0 (24A335)
Report Version:        12
Anonymous UUID:        E1A9B3E8-17C0-3635-633F-31D076EC9805

Sleep/Wake UUID:       35304D0D-F008-4006-A509-00833E2A38E4

Time Awake Since Boot: 170000 seconds
Time Since Wake:       805 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace OBJC, Code 1 

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	       0x1a03e2b4c __abort_with_payload + 8
1   libsystem_kernel.dylib        	       0x1a0408e84 abort_with_payload_wrapper_internal + 104
2   libsystem_kernel.dylib        	       0x1a0408e1c abort_with_reason + 32
3   libobjc.A.dylib               	       0x1a007c040 _objc_fatalv(unsigned long long, unsigned long long, char const*, char*) + 128
4   libobjc.A.dylib               	       0x1a007bfc0 _objc_fatal(char const*, ...) + 44
5   libobjc.A.dylib               	       0x1a00461ac lookUpImpOrForward + 836
6   libobjc.A.dylib               	       0x1a0045b84 _objc_msgSend_uncached + 68
7   CoreAutoLayout                	       0x1a9650f28 -[NSLayoutAnchor dealloc] + 36
8   CoreAutoLayout                	       0x1a9649e54 -[NSLayoutConstraint dealloc] + 72
9   CoreFoundation                	       0x1a049ff40 __RELEASE_OBJECTS_IN_THE_ARRAY__ + 116
10  CoreFoundation                	       0x1a049fe7c -[__NSArrayM dealloc] + 148
11  AppKit                        	       0x1a405af44 NSViewSetAutoresizingConstraints + 296
12  AppKit                        	       0x1a4daf658 -[NSView(NSConstraintBasedLayoutInternal) _invalidateAutoresizingConstraints] + 40
13  AppKit                        	       0x1a405b9f8 -[NSView _setSuperview:] + 444
14  AppKit                        	       0x1a407ae58 -[NSView removeFromSuperview] + 156
15  AppKit                        	       0x1a40d89bc -[NSView removeFromSuperviewWithoutNeedingDisplay] + 44
16  AppKit                        	       0x1a407f564 -[NSView _finalize] + 680
17  AppKit                        	       0x1a407f1a4 -[NSView dealloc] + 128
18  AppKit                        	       0x1a42b2b78 -[NSFrameView dealloc] + 164
19  AppKit                        	       0x1a42b2ac4 -[NSTitledFrame dealloc] + 72
20  AppKit                        	       0x1a42b2a40 -[NSThemeFrame dealloc] + 684
21  Foundation                    	       0x1a1f3cc94 _NSKVOPerformWithDeallocatingObservable + 180
22  Foundation                    	       0x1a16cd3fc NSKVODeallocate + 180
23  AppKit                        	       0x1a4233054 -[NSWindow dealloc] + 1024
24  libqcocoa.dylib               	       0x108bfeb84 0x108bd4000 + 174980
25  libobjc.A.dylib               	       0x1a0048138 objc_autoreleasePoolPop + 260
26  AppKit                        	       0x1a4965eac -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 2468
27  libqcocoa.dylib               	       0x108c009c4 0x108bd4000 + 182724
28  libFreeCADGui.dylib           	       0x10490cf7c Gui::Dialog::DlgAddPropertyVarSet::onEditFinished() + 20
29  libQt5Core.5.15.13.dylib      	       0x1042d9e84 0x104194000 + 1334916
30  libQt5Core.5.15.13.dylib      	       0x1042bd094 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
31  libQt5Core.5.15.13.dylib      	       0x1042bd094 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
32  libQt5Core.5.15.13.dylib      	       0x1042bd094 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
33  CoreFoundation                	       0x1a04fe3a8 __CFRunLoopDoSource0 + 176
34  CoreFoundation                	       0x1a04fe10c __CFRunLoopDoSources0 + 244
35  CoreFoundation                	       0x1a04fccf8 __CFRunLoopRun + 840
36  CoreFoundation                	       0x1a04fc334 CFRunLoopRunSpecific + 572
37  HIToolbox                     	       0x1ab9340cc RunCurrentEventLoopInMode + 292
38  HIToolbox                     	       0x1ab939d1c ReceiveNextEventCommon + 220
39  HIToolbox                     	       0x1ab93a020 _BlockUntilNextEventMatchingListInModeWithFilter + 76
40  AppKit                        	       0x1a403fa70 _DPSNextEvent + 660
41  AppKit                        	       0x1a49657b8 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 688
42  AppKit                        	       0x1a4032b7c -[NSApplication run] + 480
43  libqcocoa.dylib               	       0x108c004d0 0x108bd4000 + 181456
44  freecad                       	       0x1022bb184 main + 5128
45  dyld                          	       0x1a0094274 start + 2840
```

```
OS: macOS 15.0
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38827 (Git)
Build type: Release
Branch: main
Hash: abd5c3ca31539f8ed37ac3dec2b9abd252a5310d
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * Curves 0.6.47
```

pieterhijma, 2024-10-04:
I can confirm the crash. Clicking the dropdown triangle also triggers the two dialogs on Linux, although it doesn't lead to a crash there.

<details>
<summary>stacktrace:</summary>
<pre>
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x3)
  * frame #0: 0x000000010a7fbaf8 libqcocoa.dylib`___lldb_unnamed_symbol3680 + 292
    frame #1: 0x000000010a7fb614 libqcocoa.dylib`___lldb_unnamed_symbol3677 + 644
    frame #2: 0x00000001098a6b88 libQt5Core.5.dylib`QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 312
    frame #3: 0x00000001020cb5e4 libQt5Widgets.5.dylib`QDialog::exec() + 220
    frame #4: 0x00000001020f01f0 libQt5Widgets.5.dylib`___lldb_unnamed_symbol24136 + 248
    frame #5: 0x0000000104f30b90 libFreeCADGui.dylib`Gui::Dialog::DlgAddPropertyVarSet::checkName(this=0x0000600002b9c000) at DlgAddPropertyVarSet.cpp:347:9
    frame #6: 0x0000000104f2ea20 libFreeCADGui.dylib`Gui::Dialog::DlgAddPropertyVarSet::onEditFinished(this=0x0000600002b9c000) at DlgAddPropertyVarSet.cpp:410:9
    frame #7: 0x0000000104f35034 libFreeCADGui.dylib`QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (Gui::Dialog::DlgAddPropertyVarSet::*)()>::call(f=(libFreeCADGui.dylib`Gui::Dialog::DlgAddPropertyVarSet::onEditFinished() at DlgAddPropertyVarSet.cpp:399), o=0x0000600002b9c000, arg=0x000000016fdfaf50) at qobjectdefs_impl.h:152:13
    frame #8: 0x0000000104f34f78 libFreeCADGui.dylib`void QtPrivate::FunctionPointer<void (Gui::Dialog::DlgAddPropertyVarSet::*)()>::call<QtPrivate::List<>, void>(f=(libFreeCADGui.dylib`Gui::Dialog::DlgAddPropertyVarSet::onEditFinished() at DlgAddPropertyVarSet.cpp:399), o=0x0000600002b9c000, arg=0x000000016fdfaf50) at qobjectdefs_impl.h:185:13
    frame #9: 0x0000000104f34e80 libFreeCADGui.dylib`QtPrivate::QSlotObject<void (Gui::Dialog::DlgAddPropertyVarSet::*)(), QtPrivate::List<>, void>::impl(which=1, this_=0x00006000013d9220, r=0x0000600002b9c000, a=0x000000016fdfaf50, ret=0x0000000000000000) at qobjectdefs_impl.h:418:17
    frame #10: 0x00000001098c5dfc libQt5Core.5.dylib`___lldb_unnamed_symbol16968 + 820
    frame #11: 0x0000000104f2e9f8 libFreeCADGui.dylib`Gui::Dialog::EditFinishedComboBox::editFinished(this=0x0000600002b9c068) at moc_DlgAddPropertyVarSet.cpp:132:5
    frame #12: 0x0000000104f31fb4 libFreeCADGui.dylib`Gui::Dialog::EditFinishedComboBox::focusOutEvent(this=0x0000600002b9c068, event=0x000000016fdfb5f0) at DlgAddPropertyVarSet.h:53:16
    frame #13: 0x0000000101fb9f4c libQt5Widgets.5.dylib`QWidget::event(QEvent*) + 104
    frame #14: 0x0000000102032f98 libQt5Widgets.5.dylib`QComboBox::event(QEvent*) + 104
    frame #15: 0x0000000101f94974 libQt5Widgets.5.dylib`QApplicationPrivate::notify_helper(QObject*, QEvent*) + 232
    frame #16: 0x0000000101f957ac libQt5Widgets.5.dylib`QApplication::notify(QObject*, QEvent*) + 472
    frame #17: 0x0000000104ce5e48 libFreeCADGui.dylib`Gui::GUIApplication::notify(this=0x000000016fdfe708, receiver=0x0000600002b9c068, event=0x000000016fdfb5f0) at GuiApplication.cpp:93:34
    frame #18: 0x00000001098a8c74 libQt5Core.5.dylib`QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
    frame #19: 0x0000000101f97158 libQt5Widgets.5.dylib`QApplicationPrivate::openPopup(QWidget*) + 176
    frame #20: 0x0000000101fb95b8 libQt5Widgets.5.dylib`QWidgetPrivate::show_helper() + 332
    frame #21: 0x0000000101fb9da0 libQt5Widgets.5.dylib`QWidgetPrivate::setVisible(bool) + 568
    frame #22: 0x0000000102032640 libQt5Widgets.5.dylib`QComboBox::showPopup() + 2136
    frame #23: 0x00000001020330ec libQt5Widgets.5.dylib`___lldb_unnamed_symbol17038 + 228
    frame #24: 0x0000000101fb9f4c libQt5Widgets.5.dylib`QWidget::event(QEvent*) + 104
    frame #25: 0x0000000102032f98 libQt5Widgets.5.dylib`QComboBox::event(QEvent*) + 104
    frame #26: 0x0000000101f94974 libQt5Widgets.5.dylib`QApplicationPrivate::notify_helper(QObject*, QEvent*) + 232
    frame #27: 0x0000000101f966e4 libQt5Widgets.5.dylib`QApplication::notify(QObject*, QEvent*) + 4368
    frame #28: 0x0000000104ce5e48 libFreeCADGui.dylib`Gui::GUIApplication::notify(this=0x000000016fdfe708, receiver=0x0000600002b9c068, event=0x000000016fdfc090) at GuiApplication.cpp:93:34
    frame #29: 0x00000001098a8c74 libQt5Core.5.dylib`QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
    frame #30: 0x0000000101f950a0 libQt5Widgets.5.dylib`QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) + 672
    frame #31: 0x0000000101fcc984 libQt5Widgets.5.dylib`___lldb_unnamed_symbol13560 + 876
    frame #32: 0x0000000101fcbffc libQt5Widgets.5.dylib`___lldb_unnamed_symbol13554 + 248
    frame #33: 0x0000000101f94974 libQt5Widgets.5.dylib`QApplicationPrivate::notify_helper(QObject*, QEvent*) + 232
    frame #34: 0x0000000101f957ac libQt5Widgets.5.dylib`QApplication::notify(QObject*, QEvent*) + 472
    frame #35: 0x0000000104ce5e48 libFreeCADGui.dylib`Gui::GUIApplication::notify(this=0x000000016fdfe708, receiver=0x0000600003783c60, event=0x000000016fdfc7f0) at GuiApplication.cpp:93:34
    frame #36: 0x00000001098a8c74 libQt5Core.5.dylib`QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
    frame #37: 0x00000001023ea31c libQt5Gui.5.dylib`QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) + 1236
[... cut ...]
   350 	                                          "underscore, and must not start with a digit."));
</pre>
</details>

My guess is that triggering the message box multiple times leads to a crash on mac os, because Qt shares some memory between the two message boxes. Closing one deletes the memory, closing the second attempts to delete the memory as well but the memory is not accessible anymore leading to a segfault.

It's probably best and easiest to prevent to open multiple message boxes. I'm working on it.

closed 2024-10-11 by chennes