FreeCAD/FreeCAD #16790: additive loft to bend surface worked in ver 0.21.2 but in ver 1.0.0 rc it doesn't

flexbex, 2024-09-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I try to connect a square sketch to a bend surface. in older versions it worked. But in the new one I get an error on it. It fails with brep_api not done?

[image]

and in ver 0.21.2 it works without problems

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.13
  * kicadStepUpMod 10.19.5
  * Manipulator 1.5.7
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2024-09-24:
Possible fix in https://github.com/FreeCAD/FreeCAD/pull/16791

Syres916, 2024-09-24:
I think it was fixed before today as it works fine using 1.0.0RC2 as well as latest 1.1.0dev:

[image]

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Installed mods: 
  * FreeCAD-themes 2024.9.14
  * Silk 0.1.5
  * sheetmetal 0.4.24
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.27.5
  * 3dfindit-freecad-integration 1.2.0
  * fasteners 0.5.29
  * freecad.gears 1.2.0
  * Curves 0.6.45
  * A2plus 0.4.67

OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38835 (Git)
Build type: Release
Branch: main
Hash: 256ad7a01aa2753ef7b8925ede93dd9448d80a82
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Installed mods: 
  * FreeCAD-themes 2024.9.14
  * Silk 0.1.5
  * sheetmetal 0.4.24
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.27.5
  * 3dfindit-freecad-integration 1.2.0
  * fasteners 0.5.29
  * freecad.gears 1.2.0
  * Curves 0.6.45
  * A2plus 0.4.67

```

@flexbex you can download RC2 from https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/1.0rc2

luzpaz, 2024-09-25:
Closing. Please re-open if you can reproduce on latest weekly (not sure this made it into RC2)

closed 2024-09-25 by luzpaz