FreeCAD/FreeCAD #16792: Gui: Black square over menu bar since "Force OpenGL..." #16377

jffmichi, 2024-09-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Since the mentioned PR #16377 there is a black square above the menu bar. I can still click through the square and access the "File" and "Edit" entries. The square just doesn't look nice... :)

### Full version info

```shell
OS: Arch Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38742 (Git)
Build type: Release
Branch: main
Hash: b6599c5315ba557371dbdacca0b9cf35f5b7f7f2
Python 3.12.6, Qt 6.7.2, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.2.dev
Locale: English/United States (en_US) [ OS: German/Germany (de_DE) ]
Stylesheet/Theme/QtStyle: unset/Classic/breeze
Installed mods: 
  * kicadStepUpMod 11.2.7
  * dodo 1.0.1
  * sheetmetal 0.4.24
  * fcgear 1.0.0
  * fasteners 0.5.27
  * Assembly4 0.50.15
  * Curves 0.6.45
  * lattice2 1.0.0
```

### Subproject(s) affected?

None

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-09-24:
While this is a duplicate of https://github.com/FreeCAD/FreeCAD/issues/16491 , I believe it's the first time there's been a mention of it being a regression due to a PR.

Syres916, 2024-09-25:
I wondered why I didn't see this when testing on EndeavorOS and it's because the use of any stylesheet that I had to hand didn't have the issue and I didn't go back to `FreeCAD Classic` (no stylesheet) which is where it can be seen so whether you can use `FreeCAD Light` but with the _majority_ of `FreeCAD Classic` config as a workaround??

@luzpaz this is likely IMHO to have to go in `Known Issues` unless @kadet1090 has any other ideas.

Syres916, 2024-09-26:
I believe there has been a similar issue in the past which was fixed by moving the widget to X-100 therefore off the screen, now whether that applies here I'll have to do some testing on EndeavourOS and report back.

closed 2024-09-28 by adrianinsaval