FreeCAD/FreeCAD #16793: Selecting Help -> Help (F1) causes FreeCad to Crash

Robert-Botron, 2024-09-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When selecting Help -> Help (F1) Freecad Crashes.  I was in Part Design Workbench looking up information on how to create a VarSet.  It never brings up help, but just crashes.

Does something need to be specifically configured for Help to work? It does have Internet access

### Full version info

```shell
OS: Fedora Linux 40 (Workstation Edition) (GNOME/gnome-xorg)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Defeaturing 1.2.0
  * Assembly4 0.12.5
  * kicadStepUpMod 10.17.1
  * Lithophane
  * A2plus 0.4.60h
  * Help 0.4.60h
```

### Subproject(s) affected?

None

### Anything else?

/tmp/.mount_FreeCaUMEJZs/AppRun: line 41: 2187189 Aborted                 (core dumped) "${MAIN}" "$@"

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-09-24:
What's your setting in Edit --> Preferences --> General --> Help --> Display ?

AIRCAP, 2024-09-24:
Cannot reproduce with the same app image. For me it opens the help page in system browser, ubuntu 20.04 with XFCE

maxwxyz, 2024-09-25:
I cannot reproduce on Windows

luzpaz, 2024-09-25:
Can't reproduce on  
```shell
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38827 (Git) AppImage
Build type: Release
Branch: main
Hash: abd5c3ca31539f8ed37ac3dec2b9abd252a5310d
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
```

mwganson, 2024-09-28:
I cannot reproduce.  In an Ubuntu VM I have QtWebEngineWidgets installed and it brings up help in the start app.  In Windows, where I do not have it, the help opens in my default browser.  This module was removed recently from the packaging, not sure why, or if it's related to the crash.  It is used in the Help.py file,  src/Mod/Help/Help.py

In the python console, enter:

`from PySide import QtWebEngineWidgets`

Do you get an error message?

I suspect there is an issue bringing up the system default browser.

azuk, 2024-10-03:
It looks like the reporter has Help addon installed. An old version of it. According to its documentation, it is obsolete with "0.22" as that is supposed to have Help's functionality built-in.

I actually had Help addon installed as well. I couldn't reproduce a crash, but when I tried Help -> Help (F1), I got:

```
<class 'ModuleNotFoundError'>: No module named 'WebGui'
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/foo/.local/share/FreeCAD/Mod/Help/./Help.py", line 108, in show
    show_tab(html, baseurl, title, view)
  File "/home/foo/.local/share/FreeCAD/Mod/Help/./Help.py", line 222, in show_tab
    import WebGui
  File "/usr/lib/python3/dist-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'ModuleNotFoundError'>: No module named 'WebGui'
```

...and no help was opened. After I removed Help addon, Help -> Help (F1) works.

Now, interestingly, when I tried to uninstall Help addon through the addon manager, FreeCAD crashed (backtrace and other messages at the end). The crash was reproducable with Help but didn't happen when I tried to uninstall another addon. I had to remove Help by removing its directory.

```
OS: Ubuntu 24.04.1 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: recent FreeCAD-1-0 branch build, see hash
Build type: Debug
Branch: FreeCAD-1-0
Hash: 2b5dca48a50fd8e8cf2531465f40796c1c622ab2
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.8.1.dev
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Help 1.0.3
  * Render 2024.9.25
  * Curves 0.6.48
  * Assembly4 0.50.15
  * fasteners 0.5.29
  * Assembly3 0.12.2
  * CurvedShapes 1.0.10
```

When removing Help:
```
Invalid version specified for tag version in Addon Trails: 2022-01
Saved slider position at 329
QObject: Cannot create children for a parent that is in a different thread.
(Parent is Breeze::WidgetStateEngine(0x5765e0ba2fc0), parent's thread is QThread(0x5765e079a6e0), current thread is)
QObject: Cannot create children for a parent that is in a different thread.
(Parent is Breeze::WidgetStateEngine(0x5765e0ba2fc0), parent's thread is QThread(0x5765e079a6e0), current thread is)
QObject: Cannot create children for a parent that is in a different thread.
(Parent is Breeze::WidgetStateEngine(0x5765e0b9dad0), parent's thread is QThread(0x5765e079a6e0), current thread is)
QObject: Cannot create children for a parent that is in a different thread.
(Parent is Breeze::WidgetStateEngine(0x5765e0b9dad0), parent's thread is QThread(0x5765e079a6e0), current thread is)
QObject: Cannot create children for a parent that is in a different thread.
(Parent is Breeze::WidgetStateEngine(0x5765e0ba2fc0), parent's thread is QThread(0x5765e079a6e0), current thread is)
QObject: Cannot create children for a parent that is in a different thread.
(Parent is Breeze::WidgetStateEngine(0x5765e0b9dad0), parent's thread is QThread(0x5765e079a6e0), current thread is)
QObject: Cannot create children for a parent that is in a different thread.
(Parent is Breeze::WidgetStateEngine(0x5765e0b9dad0), parent's thread is QThread(0x5765e079a6e0), current thread is)
QObject: Cannot create children for a parent that is in a different thread.
(Parent is Breeze::WidgetStateEngine(0x5765e0ba2fc0), parent's thread is QThread(0x5765e079a6e0), current thread is)
QObject::installEventFilter(): Cannot filter events for objects in a different thread.
QObject: Cannot create children for a parent that is in a different thread.
(Parent is Breeze::WidgetStateEngine(0x5765e0ba2fc0), parent's thread is QThread(0x5765e079a6e0), current thread is)
QObject: Cannot create children for a parent that is in a different thread.
(Parent is Breeze::WidgetStateEngine(0x5765e0ba2fc0), parent's thread is QThread(0x5765e079a6e0), current thread is)
QBasicTimer::stop: Failed. Possibly trying to stop from a different thread
QBasicTimer::stop: Failed. Possibly trying to stop from a different thread
QBasicTimer::stop: Failed. Possibly trying to stop from a different thread
[... 58 lines cut ...]
#51  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(PyObject_Vectorcall+0x5c) [0xe1431782afc]
#52  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(_PyEval_EvalFrameDefault+0x40ee) [0xe143171d5ee]
#53  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(+0x18616a) [0xe143178616a]
#54  0xe14325e4b5d in PySide::SignalManager::callPythonMetaMethod(QMetaMethod const&, void**, _object*, bool) from d
#55  0xe14325e9ab8 in PySide::SignalManager::qt_metacall(QObject*, QMetaObject::Call, int, void**) from /lib/x86_648
#56  0xe1430306343 in QObject::event(QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x293
#57  0xe1430f6bd45 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widget5
#58  0xe1434ec554a in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/local/freecad/lib/libFreeCADGui.so+08
#59  0xe14302d8118 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so8
#60  0xe14302db94b in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) from /lib/x86_64-linuxb
#61  /lib/x86_64-linux-gnu/libQt5Core.so.5(+0x335c0f) [0xe1430335c0f]
#62  /lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x5d5b5) [0xe142e5145b5]
#63  /lib/x86_64-linux-gnu/libglib-2.0.so.0(+0xbc717) [0xe142e573717]
[... cut ...]
#68  /usr/local/freecad/lib/libFreeCADGui.so(+0x1316bf7) [0xe1434d16bf7]
#69  /usr/local/freecad/lib/libFreeCADGui.so(+0x1316f50) [0xe1434d16f50]
#70  0xe1434d1732e in Gui::Application::runApplication() from /usr/local/freecad/lib/libFreeCADGui.so+0x244
#71  /usr/local/freecad/bin/FreeCAD(+0x26cc2) [0x5765df045cc2]
#72  /lib/x86_64-linux-gnu/libc.so.6(+0x2a1ca) [0xe142f82a1ca]
#73  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0xe142f82a28b]
#74  /usr/local/freecad/bin/FreeCAD(+0x25de5) [0x5765df044de5]
```

yorikvanhavre, 2024-10-04:
The addon uninstall error might be something for @chennes to look at?

Good point about the Help addon, though. In BIM, there is now a check at FreeCAD start, that renames the BIM addon to something else if the built-in BIM is there. But that was mostly so 0.21 users can still use the BIM addon. Here I think we can do simpler, if someone is using both 1.0 and 0.21, I think we could safely assume they won't need the 0.21 help anymore and remove it...