FreeCAD/FreeCAD #16796: extrude is not cut completly (Part workbench)

JensJackowski, 2024-09-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I made an extrude of text in draft/part workbench and try to cut it out. Text containing the swedish letter Å is not cut out.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Swedish/Sweden (sv_SE)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

Part

### Anything else?

[image]

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-09-25:
The letter Å results in two overlapping solids. This causes the cut operation to fail. You need to [Part_Fuse](https://wiki.freecad.org/Part_Fuse) the extrusion and cut with the result of that tool.

closed 2024-09-25 by Roy-043

JensJackowski, 2024-09-25:
Thank you, that works.

It helped also to simply change the used font. It seems to be a bit of unclean design of the font designer. Normaly the 0 has to be above the A so that it will not generate multiplical solides.