FreeCAD/FreeCAD #16798: [UI] Regression (very recent) using the same config files, build 38794 opens with Status Bar, all previously selected toolbars and Report View alongside Python Console, build 38330 all disabled

Syres916, 2024-09-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Regression (very recent) using the same config files, build 38794 opens with Status Bar, all previously selected toolbars and Report View alongside Python Console, using build 38330 and later all are disabled on starting FreeCAD.

### Full version info

```shell
Good:

OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git) AppImage
Build type: Release
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Installed mods: 
  * FreeCAD-themes 2024.9.14
  * Silk 0.1.5
  * sheetmetal 0.4.24
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.27.5
  * 3dfindit-freecad-integration 1.2.0
  * fasteners 0.5.29
  * freecad.gears 1.2.0
  * Curves 0.6.45
  * A2plus 0.4.67

Bad:

OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38830 (Git)
Build type: Release
Branch: main
Hash: b38c42262f58cf0587b627e7a866d3756da34dc4
Python 3.12.3, Qt 5.15.13, Coin 4.0.3, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Installed mods: 
  * FreeCAD-themes 2024.9.14
  * Silk 0.1.5
  * sheetmetal 0.4.24
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.27.5
  * 3dfindit-freecad-integration 1.2.0
  * fasteners 0.5.29
  * freecad.gears 1.2.0
  * Curves 0.6.45
  * A2plus 0.4.67
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-09-24:
Git Bisect being run currently.

Syres916, 2024-09-24:
I'll narrow down the line(s) that are causing the issue but it seems to revolve around the Wayland lookup routine in MainWindow.cpp, as you can see from my system details above, I'm not running Wayland.

Syres916, 2024-09-25:
@mnesarco unfortunately I've had to revert https://github.com/FreeCAD/FreeCAD/commit/9b409da59c34ba0d302d15cd2046685091e8a06f on my local build due to this issue, at present I can't see what the root cause is but at the very least users should have the facility to change a parameter in their config files to stop this extra code from activating if a solution to cover all major distros can't be found IMHO.

mnesarco, 2024-09-25:
Hello @Syres916 what error do you get? Is a crash or just splash not showing?

mnesarco, 2024-09-25:
@chennes @3x380V what is the stablished mechanism to test this kind of changes across all distros, it seems very difficult.

Also the code is too fragile, making a splash screen visible 3 seconds should not break anything. Sounds like any gui change at this point can lead to unexpected behavior in different distros.

But if there is an stablised procedure to mitigate this problems, i want to follow it.

3x380V, 2024-09-25:
@mnesarco, there is not any. I have burnt my fingers several times, when people started reporting issues on another platforms.
I was proposing making CI results downloadable, which should be doable according to @adrianinsaval. I have not created issue for it though.

@Syres916, I hit this kind of behavior too when playing with splash screen. It pops up once you touch this sequence:
https://github.com/FreeCAD/FreeCAD/blob/54c278c1235435f1c5898334b98b103225827bd7/src/Gui/StartupProcess.cpp#L449-L452
So it seems it is limited only to Wayland, but happens on X as well and it is therefore probably platform independent.

luzpaz, 2024-09-25:
> I was proposing making CI results downloadable  

I am for this

mnesarco, 2024-09-26:
Hello @chennes @Syres916 what is the procedure to revert this? a new PR?

Syres916, 2024-09-26:
@mnesarco yep a new PR gives traceability for others IMHO

3x380V, 2024-09-28:
@mnesarco, could you, by a chance, give #16334 a try? You'll need also enable messages in user.cfg:
`<FCBool Name="ShowSplasherMessages" Value="1"/>`
It is also possible you'll need similar fix as described in https://github.com/FreeCAD/FreeCAD/pull/16869#pullrequestreview-2335400809

3x380V, 2024-09-29:
Temporarily added @mnesarco's updated revert as a part of #16334 to make testing easier.

mnesarco, 2024-09-29:
> @mnesarco, could you, by a chance, give #16334 a try? You'll need also enable messages in user.cfg: `<FCBool Name="ShowSplasherMessages" Value="1"/>` It is also possible you'll need similar fix as described in [#16869 (review)](https://github.com/FreeCAD/FreeCAD/pull/16869#pullrequestreview-2335400809)

@3x380V I have cloned, compiled and tested. It works on my side, the splash is visible for a very short time but nothing else get broken. The splash is visible for too short in my opinion but a little better than before. Here is the screencast:

https://github.com/user-attachments/assets/9a171d2e-0a84-4a52-b1b1-43f1851bae23

To be honest, I have learned that the whole thing is a house of cards. There are no quick workarounds.

3x380V, 2024-09-29:
@mnesarco, it seems you do not have messages enabled, see my previous post how to do that. Also, I'll be glad, if you could review #16438.

mnesarco, 2024-09-30:
> @mnesarco, it seems you do not have messages enabled, see my previous post how to do that. Also, I'll be glad, if you could review #16438.

Tested with `<FCBool Name="ShowSplasherMessages" Value="1"/>`: 

https://github.com/user-attachments/assets/11126dab-9ab3-4157-8963-3de47edc800b

I don't know what is the purpose of those messages there, they are unreadable anyway.

mnesarco, 2024-09-30:
@3x380V I am not the right person to review https://github.com/FreeCAD/FreeCAD/pull/16438, i do not have a wayland environment neither other distros to test it properly.

3x380V, 2024-09-30:
@mnesarco, the whole purpose of splash screen is to show something to the user while application is initializing. At the time application begins to run, it is unknown how long will initialization take. Later on user can decide to disable splash entirely. As for messages, even when they are unreadable [*] there is something shown in case app gets stuck on start. Do not look at this with optics of your fast machine. Others may run slower computers or workbenches taking way longer to init. Thank you for testing this.

[*] in case you are talking about visual, I was asking several times how it should look like and there was no answer. I'm not a designer and I will not pretend I'm able to come with a nice solution myself.

mnesarco, 2024-09-30:
@3x380V As I understand it, the initialization process is too quick to display logs. In my opinion, the splash screen isn't designed to serve as an error or log display. I believe its primary purpose is aesthetic. As a backend developer, I'm used to seeing initialization messages and errors in a log file or console. However, I understand that desktop applications often have different approaches. It seems I may have misunderstood the context of this issue from the start.

3x380V, 2024-09-30:
@mnesarco, FreeCAD was displaying startup messages since dawn of git history:
https://github.com/FreeCAD/FreeCAD/blob/6e2cd4e73303f817e4bf5e908abe39030b7c5fb3/src/Gui/Splashscreen.cpp#L195-L217
Note the pattern searched and note the delay added after each message printed to allow people to see it. I wanted that delay go away, so implemented #15435, but it did not really work as main thread is doing initialization, only main thread can interact with gui and event loop is not running, so QTimers are out of scope. Thus only scrolling remains, although very fast, so barely readable.
I would expect most users start FreeCAD by clicking on the icon, so there is no console and it is not immediately clear where logs are stored, if there are any at all. That's why I think there is still some value in seeing startup messages as we could before #16071 was merged.
Also note, that #16334 still leaves messages disabled by default. It just implements scrolling and eliminating delay. How should that look visually (I found it ugly too) is beyond my scope, I just provided "backend". Shortly, I do not consider it a solution, just foundation to future work.

luzpaz, 2024-10-11:
What's to be done here ?

adrianinsaval, 2024-10-12:
> Also note, that #16334 still leaves messages disabled by default. It just implements scrolling and eliminating delay. How should that look visually (I found it ugly too) is beyond my scope, I just provided "backend". Shortly, I do not consider it a solution, just foundation to future work.

maybe instead of displaying messages we should just display a progress bar?

3x380V, 2024-10-12:
@adrianinsaval, this progress bar idea keeps popping around in regular intervals :) You cannot have regular progress bar as it needs to know number of initialization steps in advance, which you do not know. Or do you mean indeterminate (marquee) progress bar? This one is possible, although I've never understood why is such a thing called progress bar in the first place.

3x380V, 2024-10-12:
> What's to be done here ?

That's easy: review #16438, merge it to the master branch, wait for eventual bug reports, repeat the same with #16334. Now you should be in the same state at least regarding code path executed. Then implement indeterminate progress bar in `SplashScreen::drawContents` and add config entry selecting between no progress, messages and bar. Wait for bug reports and backport all that into stable branch.

closed 2025-08-10 by github-actions[bot]