FreeCAD/FreeCAD #16801: [Materials][CAM] new Machinability requires coding changes for every Tool-Material & is missing Fz(chipload) data

spanner888, 2024-09-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

new Machinability Materials will require coding changes for every Tool-Material & is missing Fz(chipload) data
The new Machinability Materials Model provides basis of calculating cutting RPM and power and includes a very coherent dataset and example calculations.

It includes data for two values to aid calculation of various cutting power calculations and two values of Vc, one each for Tool Material of HSS and Carbide.

Those two Tool materials are an excelent start for a generic tool based Speeds and Feeds calculations. However in future it is reasonable to expect at least the addition of tool coatings and no doubt requests to allow specific supplier Tools.

At present any changes to Tool materials, require changes to both the Machinability model, materials and any code that uses the associated values.

In addition current model is also missing fz (chipload) data.

The 'extended' machinability model proposed below suggests ways to solve both these issues and an alternate way to store the fz data, instead of just the raw value for each discrete value of Tool-Diameter.

Example code is buried within the just released CamScripts addon and includes a single example. See the README.md at https://github.com/spanner888/CamScripts

Proposed model
```
Model:
  Name: 'MachinabilityVcFzExtensions'
  UUID: 'b3aab784-ee76-4072-94b9-e028e27264be'
  URL: 'https://en.wikipedia.org/wiki/Speeds_and_feeds'
  Description: >-
    Model for calculating feeds and speeds on a milling machine or lathe
  Inherits:
    - Machinability:
      UUID: '9d81fcb2-bf81-48e3-bb57-d45ecf380096'
  DOI: ""
  Vc:
    Type: '2DArray'
    Columns:
      VcToolMat:
        Type: 'Quantity'
        Units: '1'
        URL: ''
        Description: "Index to tool Material list"
      SurfaceSpeed:
        Type: 'Quantity'
        Units: 'm/min'
        URL: ''
        Description: ""
    URL: ''
    Description: >
      2 Dimensional array SurfaceSpeed for each Tool Material
  Fz:
    Type: '2DArray'
    Columns:
      FzToolMat:
        Type: 'Quantity'
        Units: '1'
        URL: ''
        Description: "Index to tool Material list"
      fzSquared:
        Type: 'Quantity'
        Units: '1/mm'
        URL: ''
        Description: "x^2 curve var, can be zero for linear equation"
      fzSlope:
        Type: 'Quantity'
        Units: '1'
        URL: ''
        Description: "x^1 curve var"
      fzIntercept:
        Type: 'Quantity'
        Units: 'mm'
        URL: ''
        Description: "Y axis intercept"
    URL: ''
    Description: >
      Fz parameters for each Tool material
  ToolMat:
    Type: 'List'
    URL: ''
    Description: "List of Tool Material. Vc & Fz must be in same order to match data!"
```

Example Material Card:
```
---
# File created by FreeCAD 0.22.0 Revision: 38459 (Git)
General:
  UUID: "5043f450-2158-4aa7-ba22-59f0db62e391"
  Name: "AlCastAlloyINHERITED+fz"
  License: "All rights reserved"
Inherits:
  AluminumCastAlloy:
    UUID: "048ba0ec-1fc4-4e3c-9710-bde305a1503f"
Models:
  MachinabilityVcFzExtensions:
    UUID: 'b3aab784-ee76-4072-94b9-e028e27264be'
    Vc:
      - [["0", "80.00 m/min"],
        ["1", "120.00 m/min"],
        ["2", "31.00 m/min"]]
    # Fz Only row 0=HSS & row 1=Carbide has real data ATM
    Fz:
      - [["0", "-0.00020581", "0.00861056", "-0.03485845"],
        ["1", "-0.00024088", "0.00965887", "-0.037093"],
        ["2", "0.0", "0.0", "0.0"]]
    ToolMat:
      - "HSS"
      - "Carbide"
      - "HSS coated"
      - "Carbide coated"
      - "Ceramic"
```

Example output:
```
ToolBit: Test_JobUtils#_F_D6_4_L50_0_roughing_HRangles , has HelixAngle, using value: 17.2 deg
ToolBit: Test_JobUtils#_F_D6_4_L50_0_roughing_HRangles , has RakeAngle, using value: 29.9 deg
material : AlCastAlloyINHERITED+fz

mat inf:  Material/Machining/AlCastAlloyINHERITED+fz.FCMat 
	 Custom /home/spanner888/Documents/_APPS_lappy/FC_wkly-38827/appd_mlappy/Mod/CamScripts/freecad/cam_scripts/cutting_tool_data/Resources
parent inf: Machining/AluminumCastAlloy.FCMat System /tmp/.mount_FC_wklZxQ9tw/usr/share/Mod/Material/Resources/Materials
Vc array data [[0.0, 1333.3333333333335 mm/s], [1.0, 2000.0000000000002 mm/s], [2.0, 516.6666666666667 mm/s]]
Vc for Tool Mat: HSS  is:  1333.3333333333335 mm/s 1333.3333333333335 mm/s

Fz.Array data:
	 HSS [0.0, -0.00020581, 0.00861056, -0.03485845]
	 Carbide [1.0, -0.00024088, 0.00965887, -0.037093]
	 HSS coated [2.0, 0.0, 0.0, 0.0]
Calculated Fz for Tool Mat: HSS  is:  0.011819156400000003 mm

Calculated RPM = 3979
electrical spindle power  0.153 kW
Mc cutting torque 0.31204 Nm
vf  141 mm/min
mrr  45 mm^3/s
```

@jbaehr @sliptonic 

cc @davesrocketshop 

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38827 (Git) AppImage
Build type: Release
Branch: main
Hash: abd5c3ca31539f8ed37ac3dec2b9abd252a5310d
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/Australia (en_AU)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * CamScripts
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

sliptonic, 2024-09-25:
I have always understood that chipload is an attribute of the tool geometry.  It's usually derived experimentally and supplied by the tool manufacturer.  It seems odd to make it an attribute of the material. 
Do I misunderstand?

jbaehr, 2024-09-25:
For me it looks odd to have the chip load as a property of the tool bit. Sure, there are some geometrical limits on the max chip load before the chip evacuation channel is jammed, but (unless you machine foam, perhaps) you will never reach this limit. So for practical means the max chipload is more about usage than geometry: Finishing passes use smaller chiploads than roughing. Machining MDF allows larger chiploads than machining aluminium, etc.

For a professional environment @sliptonic's argument from above sounds plausible, though. There you may have dedicated bits for each task and material thus it's not that much of a problem to attach this property directly to the tool bit.  
On the other hand, I as a hobbyist do not have such a variety available. I do reuse the same bits for slotting, roughing and finishing. I do reuse the same bits for aluminium and wood. And I assume that there are more hobbyists/amateurs/makers using FreeCAD for CAM than professionals. I further assume that hobbyists will benefit more from additional guidance than trained personell. Here the inconvenience starts. Either I duplicate my tool bits in the library for each task and material -- or I just don't use the chipload property there.

So I think it's best to store only properties in tool bit library that are only dependent on the bit itself. That is the cutting material (HSS, Carbide, Stellite, Ceramics, Dimond, ...) and possible coatings, its geometry (dimensions, rake- and helix angels, etc.) and limitations such as the max permissible PRM. 
The chipload does not only depend on the bit, but also on the kind of operation and the machined material.

--------

I think it's a generally good idea to put the chipload together with the surface speed into the material cards. I just don't know yet *how* -- I'm not convinced by the current proposal for a few reasons:
- Due to the indirection of the cutting material it's hard to maintain, i.e. it's easy to make mistakes that are not easy to spot.
- Storing the coefficients of a polynomial is hard to understand and replicate, at least without additional tooling. Much easier would be just provide a list of explicit samples, that can then be used as supporting points for whatever interpolation gets implemented in FreeCAD. I as a user and card author can then just copy tables from other resources as-is. And FreeCAD is free to change/improve the interpolation algorithm without breaking existing cards.
- It does not take the various kinds of operations into account. E.g. you cannot express different surface speeds and chiploads for roughing and finishing. Or drilling. Or (slit-)sawing. 
- Adding values for new cutting materials requires changing existing cards. According to the Open/Close-Principle it would be nice if one could just "add" without "change". I want to be able to extend the materials provided by the installation without having to touch them and so either lose my additions on the next update or don't get potential enhancements from the next update. If we can find a way to make this work by composition instead, that would be clearly the preferred solution.

I've already sketched some ideas in [this comment](https://github.com/FreeCAD/FreeCAD/pull/15910#issuecomment-2314463872) but I'm not convinced by this, either. While it does offer different values for different operations, the cutting material is not data-driven. Having a new model for a new operation seems fine. Having a new model for a new cutting material is not ideal (in the same way as the static list, i.e. the current "HSS" and "Carbide", in the tool bit library is not ideal). And the fact that all this multiplies makes me shiver...

--------

A word on "Fz" and "Chipload". @spanner888 seems to use these interchangeable in this issue's description, but I think that's not the same. For me, "Fz" is the "feed per tooth", i.e. a measure of how fast your tool moves in X/Y/Z per spindle revolution and tooth. "Chipload" is a measure of how much material is removed per spindle revolution and tooth. For small radial depth of cut (<< 50% of the tool diameter), the chipload is smaller than the feed per tooth.
Example: For a profiling operation with a 3mm endmill I may choose a radial depth of cut of 0.75mm (20% of the diameter). If I aim for a max chipload of 0.025mm in aluminium, I can work with a feed per tooth of 0.029mm. For shallower finishing operations the effect is even greater. For a sample implementation see the script provided in my [Wood Cards MR](), for more details see e.g. [this article on Chip Thinning](https://www.machiningdoctor.com/calculators/chip-thinning-calculator/).

spanner888, 2024-09-27:
> A word on "Fz" and "Chipload", yes you are correct @jbaehr, a disapointing mistake from me, I edited with strikethrough.

spanner888, 2024-09-27:
I have been writing positive response to all of your great comments, but getting endlessly lost in far too many words and details, which has made me realise, the thing I believe really needs discussing is:

Can we have some discussion on Goals and maybe intended stages? For example:
    Provide milling Speed, Feed, Power outputs calculated from included Machinability data.
    How will outputs appear ...in the FreeCAD Job?
    What cutting operations will be supported Contour??, Profile, Slotting, Roughing, Finishing??
    Proposed major goals/stages...or even basic 1st goal, then review future....
    Only using supplied data...

I have seen past forum comments which included from memory some ideals like simple calculator, but no real specifics and nothing recent. 

Such a discussion and hopefully agreement would help both contributors and help set user expectations.

Maybe can summarise in another github FreeCAD project.

knipknap, 2025-01-10:
@jbaehr I agree with your comments - it can be boiled down to this: Manufacturers specify an ideal chipload (which is probably more or less the same as a maximum chipload), but under the assumption that the bit is used for one particular scenario (material and operation type). Without knowing what scenario that is, the value is not useful.

In addition, there are "out of spec" chiploads which users determine from experimentation.

So maybe a good approach would be this:
- The chipload should be a tool attribute, but the tool should also have a reference to materials and and operations. This will allow to identify when a tool is used "out of spec".
- If the user assigns it to a job in an "out of spec" manner, FreeCAD can determine a "default chipload" based on the tool diameter and a material a specific constant (=chipload divisor). This is what I do in BTL's feeds&speeds calculator.

This would address the "chipload" part of the material data model and would probably cover 95% of use cases. There is still a small gap: If a tool manufacturer specifies multiple chiploads depending on material type, then the tool data model should allow for specifying that as well.

Additional note: To actually calculate feed rates, additional material-specific data is needed, in particular the min and max speeds per operation type. But if I understand it correctly, this part of the data is not subject of the discussion in this thread.