FreeCAD/FreeCAD #16802: Changes in design not visual in export file

Eggberry, 2024-09-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After making changes in a design in "Part Design" mode, and save this under a new name,  a new export to a stl file does not include the changes.  

### Full version info

```shell
OS: Slackware 15.0 x86_64 (KDE//usr/share/xsessions/plasma)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git) AppImage
Build type: Release
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Render 2024.8.4
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-25:
Too generic...
Steps to reproduce?  
Example file?

luzpaz, 2024-09-25:
@Eggberry can you zip up the file, drag/drop it into the github textbox ? This will auto-attach it to the ticket.

FlachyJoe, 2024-09-25:
Such request should go to the forum not here. Couldn't we mention the forum at issue creation?

luzpaz, 2024-09-25:
@FlachyJoe we do:  
[image]

closed 2024-09-25 by luzpaz