FreeCAD/FreeCAD #16804: FreeCAD 1.0 Crashed on Startup (Linux AppImage)

Batwam, 2024-09-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FreeCAD aborts on startup when installed as AppImage on Ubuntu 24.10. I am using x11.
The splash screen appears and things start to load but it eventually closes.

FreeCAD version: FreeCAD-0.21.2-Linux-x86_64

Couple of dependencies appear to be missing followed by an error with QtWebEngine/OpenGL. 

Seems similar to https://forum.freecad.org/viewtopic.php?t=79466 but the issue still occurs after restart

### Full version info

```shell
user@machine:~/AppImage$ ./FreeCAD-0.21.2-Linux-x86_64.AppImage 
FreeCAD 0.21.2, Libs: 0.21.2R33771 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

MESA-LOADER: failed to open iris: /usr/lib/dri/iris_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
failed to load driver: iris
MESA-LOADER: failed to open iris: /usr/lib/dri/iris_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
failed to load driver: iris
MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
WebEngineContext used before QtWebEngine::initialize() or OpenGL context creation failed.
QGLXContext: Failed to create dummy context
Failed to create OpenGL context for format QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize 24, redBufferSize -1, greenBufferSize -1, blueBufferSize -1, alphaBufferSize -1, stencilBufferSize 8, samples 0, swapBehavior QSurfaceFormat::DefaultSwapBehavior, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile) 
/tmp/.mount_FreeCAvG7tzX/AppRun: line 43: 2659372 Aborted                 (core dumped) ${MAIN} "$@"
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Krypto31337, 2024-09-25:
The easiest solution is to use Freecad 1.0rc2.

You need to use LD_PRELOAD to help the 0.21 appimage with the new Ubuntu releases.

https://forum.freecad.org/viewtopic.php?p=781110

Batwam, 2024-09-25:
Thanks, I'll the latest release candidate and revert back.

Batwam, 2024-09-25:
ok, all good now

closed 2024-09-25 by Batwam