FreeCAD/FreeCAD #16806: [BIM]  Import IFC Locked/UnLocked Mode  has different behavior of same import strategy

hoshengwei, 2024-09-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In Ifc import strategy, there provide 3 ways to initial import the ifc file

[image]

in code:
```python
strategy:  0 = only root object
           1 = only bbuilding structure,
           2 = all children
```

and then u can choosed the Locked Mode or Unlocked Mode.

In my opinion that when the same strategy is chosen, the import behavior in locked mode and unlocked mode should be the same. The only difference should be whether the IfcProject belongs to a FreeCAD document or an IFC document.

However, if you choose strategy 0 (only root object), in locked mode it will import IfcProject -> IfcSite -> IfcBuilding, whereas in unlocked mode it will only import IfcProject.

If you choose strategy 1 (only building structure), in locked mode it will import IfcProject -> IfcSite -> IfcBuilding -> IfcBuildingStorey, but in unlocked mode it will import IfcProject -> IfcSite -> IfcBuilding -> IfcBuildingStorey -> IfcSpace.

This is just a minor issue, and it doesn't affect the program's functionality. I just feel that some details could be optimized and adjusted in the future, so I'm making a note of it.

### Full version info

```shell
OS: Arch Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38831 +4 (Git)
Build type: Release
Branch: materialAddSupprtIfc
Hash: 6db83f102a6344f3cda85e76b20c47d18204a245
Python 3.12.6, Qt 6.7.2, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

semhustej, 2024-09-25:
I can confirm the behavior. There is also difference in strategy 0:

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38830 (Git)
Build type: Release
Branch: main
Hash: b38c42262f58cf0587b627e7a866d3756da34dc4
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

```

maxwxyz, 2024-09-25:
@yorikvanhavre FYI

yorikvanhavre, 2024-10-01:
> If you choose strategy 0 (only root object), in locked mode it will import IfcProject -> IfcSite -> IfcBuilding, whereas in unlocked mode it will only import IfcProject.

Indeed. The problem here, is that in locked mode, the project IS the FreeCAD document. Therefore there is no object in the document to represent the project contents. That's why I made it show the site and building. Not sure that's the best strategy, though, but I had no better idea... Using a "dummy" object seemed wrong and give even more confusion to users

> If you choose strategy 1 (only building structure), in locked mode it will import IfcProject -> IfcSite -> IfcBuilding -> IfcBuildingStorey, but in unlocked mode it will import IfcProject -> IfcSite -> IfcBuilding -> IfcBuildingStorey -> IfcSpace.

Are you sure? It's exactly the same function that runs in both cases. Would you have a test file?

hoshengwei, 2024-10-03:
> Indeed. The problem here, is that in locked mode, the project IS the FreeCAD document. Therefore there is no object in the document to represent the project contents. That's why I made it show the [...]

how about make unlock mode import the level to IfcBuilding as well ? 

> Are you sure? It's exactly the same function that runs in both cases. Would you have a test file?

This is my bad,  after I check again,  both unlock/lock mode will import  IfcSpace when choose the strategy 1

should we close the issue?

yorikvanhavre, 2024-10-03:
> how about make unlock mode import the level to IfcBuilding as well ?

Yes, but then it is identical to strategy 1. I understand strategy 0 as "I just want to have the contents of the IFC file into one, easily movable/manageable object, I don't want the details".

One thing I researched already earlier on, is to build a Coin representation of the model, and just display it in the viewport, with no object in the tree. That kind of works, but then there is nothing to double-click to expand, nothing to manipulate, and it feels very weird in FreeCAD.

> should we close the issue?

Let's leave it open a bit more... Maybe someone will come up with a better idea!