FreeCAD/FreeCAD #16808: Persistent Section Cut very slow and errorous

blubbsy, 2024-09-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hi,

i have a imported STEP assembly with originally slightly over 1000 components. I cleaned it up and reduced it to few hundred components to get rid of pins, ICs etc from the PCB. 
But when i try to make a Persistent Section Cut it takes like 6h to make the cut and then it does not cut all parts, but only randomly a 2-3 components. i have to admit I haven't counted properly how many components were cut and only checked on the biggest ones. Funny enough, all other similar features (e.g. clipping plane, Cross-Section) work like a charm and fast.
Sadly i cant share the file with you to check if it a specific problem with this assembly :(

### Full version info

```shell
1.0 RC2
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-09-25:
That's the limitation of this tool as it actually cuts the model using boolean difference with a box and needs to compute the result. On the other hand, hollow cuts like Clipping plane are indeed lightweight.

luzpaz, 2024-09-25:
> There are plans to introduce a proper filled cut tool like the one in Link Branch.

Is there a FR for this ?

FEA-eng, 2024-09-25:
> Is there a FR for this ?

Sure: #5570

blubbsy, 2024-09-26:
@FEA-eng  thanks for the response, but does this really explain why it takes hours? and why doesnt it then cut all parts in the assembly but just some? so even if it is slow because of the boolean operation, it should then actually deliver a right result.

FEA-eng, 2024-09-26:
> @FEA-eng thanks for the response, but does this really explain why it takes hours? and why doesnt it then cut all parts in the assembly but just some? so even if it is slow because of the boolean operation, it should then actually deliver a right result.

It should be quite easy to manually do what this tool does internally: https://wiki.freecad.org/Part_SectionCut#Background_Info
Then you would see what happens for your model.

FEA-eng, 2025-07-28:
Still relevant but it might be more of a feature request than a specific bug