FreeCAD/FreeCAD #16810: Core: UI: Spinboxes are too wide.

PaddleStroke, 2024-09-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In Freecad the spinboxes are annoying big. Very often I came upon this problem. If the spinbox does not have a maximum, then it is very wide. If you set a maximum of 99.9 to a spinbox, then it is reasonably sized. But if no maximum the size is so big as to accomodate a 12 digit number. Which virtually never happens so the interface is screwed for nothing.

Take the dialog placement for example. This is its current minimum size:
[image]

Which is ridiculuously large on most screens. And just because these spinboxes are so large. This dialog should look like this : 
[image]

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-09-29:
Frame titles are rendered below other elements (spinboxes and a combobox).

adrianinsaval, 2024-09-29:
is this what determines this?
https://github.com/FreeCAD/FreeCAD/blob/6e2cd4e73303f817e4bf5e908abe39030b7c5fb3/src/Gui/QuantitySpinBox.cpp#L793-L850

PaddleStroke, 2024-09-30:
Well we can try adding a * 0.5 and see what happens !

xtemp09, 2024-09-30:
> Well we can try adding a * 0.5 and see what happens !

I would take existing implementation of `minimumSizeHint()` and `sizeHint()` from somewhere else and adopt it here. =/

PaddleStroke, 2024-09-30:
I didn't meant as a solution. 😄 
Just a way to test if the size is indeed enforced here.

PaddleStroke, 2024-09-30:
Indeed changing QuantitySpinBox::sizeHint and QuantitySpinBox::minimumSizeHint() both works. Btw they are copy paste of each other.