FreeCAD/FreeCAD #16811: VarSet Non-obvious rules for the number of characters when creating a property.

kaiwas, 2024-09-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

At the time of creating the property propertyfloat, there is no way to enter more than two characters. I think that two characters are clearly not enough. You can change the property after creation, but this is an unnecessary action (for example, I can say that I often have to specify numbers over 1000)

At the time of creating a property, the length (and similar) restrictions of two characters are not, but if I specify more than nine characters, the property will take a value equal to zero.

Probably, if there is a limit of 9 characters, it should work the same everywhere.

https://github.com/user-attachments/assets/1c20d562-1834-43c1-8bf5-bac38a222acd

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git) AppImage
Build type: Release
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * PitchedRoof
  * lattice2 1.0.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-09-25:
@pieterhijma FYI

maxwxyz, 2024-11-15:
Closing this in favor or https://github.com/FreeCAD/FreeCAD/issues/17873