FreeCAD/FreeCAD #16812: Opening file with helix shows error: “No module named 'ThreadProfileCmd'”

mpfusion, 2024-09-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Opening the file [helix.tar.gz](https://github.com/user-attachments/files/17127285/helix.tar.gz) shows the following errors in the console:

````
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'ModuleNotFoundError'>: No module named 'ThreadProfileCmd'
````

Expected behaviour: No errors shown

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (X-Cinnamon/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38830 (Git)
Build type: Release
Branch: main
Hash: b38c42262f58cf0587b627e7a866d3756da34dc4
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/Ireland (en_IE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * fasteners 0.5.29
  * kicadStepUpMod 11.3.9
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-25:
An aside, 3rd party dependencies OCC and coin need to be updated.

mpfusion, 2024-09-25:
*Need to* or *may need to*? I use Debian stable. That's the versions Debian ships in the stable distribution.

luzpaz, 2024-09-25:
@mpfusion lets say highly recommended.  
There are some important fixes in Coin and OCCT.  
You may want to consider using flatpak, appimage or snap as an alternate way to test if something may be 3rd party dependency related and/or packaging related.

luzpaz, 2024-09-25:
I get a different error trying to load file:

```shell
11:59:46  Migrating Start Workbench to Start command... done.
12:01:02  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'OSError'>: Invalid project file /home/foo/Downloads/invalid_sketch_test.fcstd: basic_ios::clear: iostream error
```

```shell
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38827 (Git) AppImage
Build type: Release
Branch: main
Hash: abd5c3ca31539f8ed37ac3dec2b9abd252a5310d
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
```

mpfusion, 2024-09-25:
I was using the appimage for a few months but ran into various issues and then started to compile the most recent version, which worked much better than the appimage. But for the sake of this bug report I gave it another try and the issue is present in the appimage as well:

```
Traceback (most recent call last):
  File "/tmp/.mount_FreeCAGrfbYu/usr/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'ModuleNotFoundError'>: No module named 'ThreadProfileCmd'
```

FreeCAD_weekly-builds-38827-conda-Linux-x86_64-py311.AppImage

depthoffocus, 2024-09-26:
Isn't this just that your installation lacks the third-party ThreadProfile workbench that was used to create VThreadProfile in the design?

depthoffocus, 2024-09-27:
> You're right, after installing the workbench the error vanishes. Frankly, I wasn't aware of the fact that one needs to install the workbenches used when _creating_ the file to merely _open_ them.

I am just a passer-by but I think this varies depending on what the workbench added to the file. 

e.g. in your file, there is one of these:

```
<Property name="Proxy" type="App::PropertyPythonObject">
  <Python value="IlRocmVhZFByb2ZpbGUi" encoded="yes" module="ThreadProfileCmd" class="_ThreadProfile"/>
</Property>
```

Which looks to me like it's saying _"this object isn't completely defined here, please ask this module what this encoded data means"._

> 
> This however begs the question: how is the user supposed to know which of the hundreds of extentions to install to open a file (which they might not have created themselves).

I guess, at the moment, it would be down to the creator of the file to document (whether in some associated file or in a readme text file added into the FreeCAD project) which extensions are needed. The Document.xml and GuiDocument.xml files do make some references to the "providers" and dependencies of various elements, so I guess FreeCAD could -- at the project level -- maintain more data in the registry to match these to plugins. But it might make more sense to have some additional metadata to make it explicit.

I think you can close your own ticket if you like, but yeah, I'd first search to see if there are any issues like this and if not, create a new ticket as you are suggesting, to raise this topic.

luzpaz, 2024-09-30:
Hey @mwganson any idea how freecad should let the user know if a function is available in a 3rd party wb when they open a file that utilizes said wb? (like yours)

mwganson, 2024-09-30:
FreeCAD has no way of knowing which addons contain the missing module, but it does presumably have the object itself.  For example, the objects could all have a string property named "About" and this property could contain a string such as "Created with ThreadProfile workbench version 1.23".  If when handling the ModuleNotFound exception the object has the About property, then that property could be incorporated into the error message.

I'm not sure where in FreeCAD this error message is getting generated.  It's a generic pyException, so searching for "No module named" doesn't yield any results.

depthoffocus, 2024-10-01:
> FreeCAD has no way of knowing which addons contain the missing module, but it does presumably have the object itself. For example, the objects could all have a string property named "About" and this [...]

The FreeCAD website could maintain a URL redirector for the addons repository, too.  

So say:

`https://freecad.org/links/addon/ThreadProfile`

would effectively resolve to the relevant link in https://github.com/FreeCAD/FreeCAD-addons

And that would mean 

1) information in the error messages will stay current to wherever that workbench is actually hosted at any point in the future
2) a consistent URL scheme that a future FreeCAD could intervene in, to link a better error report directly to the module view in the addon manager
3) deprecation or vulnerability notices could be added in future

FlachyJoe, 2024-10-01:
_package.xml_ could inform about objects the addon implements so the AdonManager could be asked for an object based on its cache content.
The _Proxy_ object property helps too.
```python
>>> str(App.ActiveDocument.Line.Proxy)
'<freecad.Curves.lineFP.line object at 0x7f17916798e0>'
>>> str(App.ActiveDocument.Placment.Proxy)
'<lattice2Placement.LatticePlacement object at 0x7f179150d5b0>'
```

mwganson, 2024-10-01:
The Proxy object is what is missing because that is what the missing class/module defines, so you wouldn't be able to use it.  The property could be a string that uniquely identifies the addon in the addon manager.  In the error message FreeCAD could offer to open the addon manager to install the missing addon.  There is already a regex search box, so the property string could be some text that could be populated into that search box.

FlachyJoe, 2024-10-01:
@mwganson Yes but the error reported when opening a file which contains an object from a missing addon mention it
e.g. `<class 'ModuleNotFoundError'>: No module named 'lattice2Placement' `

mwganson, 2024-10-01:
Ideally, that module name could be used by the addon manager to present to the user the list of addons that use it.  Then there is no need for the chore of adding dynamic string properties to all of those objects.  Adding all the modules to the package.xml file would also be a chore, and not every addon that generates custom scripted objects even has a package.xml file available, for example: Honeycomb macro on the wiki.

For ThreadProfile it would be easy enough, only one module file is used: ThreadProfileCmd, but for Curves workbench, Lattice2, and others, it would be quite a lot of files to add.

Better if the addon manager can gather and store this information.  It knows the names of the files used when installing an addon, so it just needs to keep track of which .py files go with which addon.  Certain common names would be ignored, like Init.py.  It could then provide a function that could be called by the error handler to perhaps offer to open the addon manager with the addon, or addons if there are multiple options that use the same named module, selected and ready to be installed.

luzpaz, 2024-12-09:
@chennes maybe we should rename this ticket (or open a new one) as an FR that FC have the ability to detect that a missing function is from a 3rd party addon and let the user know this + from which one + and maybe a link into the addon manager to download it conveniently ?

luzpaz, 2025-01-17:
I've updated the ticket name per @mwganson's feedback.

chennes, 2025-01-17:
> It knows the names of the files used when installing an addon

It really doesn't -- it's just doing a "git clone" or unzipping an archive, it never looks at individual filenames. IMO if an addon is creating an object that requires code from that addon to work, that should be indicated in the object's properties, and that's a job for the addon author, not some external code. The information needs to be in the FCStd file itself.