FreeCAD/FreeCAD #16816: UX: Visibility icons (eyes) shown for spreadsheets and varsets

hyarion, 2024-09-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

(from a discussion on discord with Max)

There are non-functional visibility icons for both Spreadsheets and VarSets.

As these objects aren't shown in the 3d view, the visible icon doesn't make sense to me (an invisible-icon wouldn't make sense either)

[image]

I think the icon should changed to invisible Since neither visibility state is valid for this 
The icon should be removed (without changing the primary icon alignment) in my oppinion:
[image]

### Full version info

```shell
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CurvedShapes 1.0.10
  * sheetmetal 0.2.61
  * Curves 0.6.6
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-25:
This is a nice UI fix. Thanks @hyarion !

maxwxyz, 2024-09-25:
@FreeCAD/design-working-group as this is hardcoded: do you know any other object types which cannot be hidden?

pierreporte, 2024-09-25:
There are TechDraw pages. They can actually be hidden, which keep FreeCAD from displaying them when active, but it doesn’t make sense.

maxwxyz, 2024-09-25:
@FEA-eng anything in FEM? Like the solver objects or materials?

FEA-eng, 2024-09-25:
> @FEA-eng anything in FEM? Like the solver objects or materials?

Yes, anything else than constraint with symbols, meshes or results pipeline objects (so mostly materials, tools applied to meshes and element properties plus solvers and some constraints acting on whole bodies) but regardless of the visibility icons, it has always been possible to hide them which only grays out their icons and names in tree view.

hyarion, 2024-09-25:
Ok, this quickly spiraled away in scope!

I thought I would be able to fix it in a quick change for 1.0 like this draft PR https://github.com/FreeCAD/FreeCAD/pull/16817 but it is becoming clear that maybe we need to find a better way.

To solve this in a good way I think we need to:
1. Make `DocumentObject` keep a boolean/property/state which declares if it can be hidden/shown (`isVisualObject` <-- placeholder name).
2. The eye icon and context menu visibility options should only be shown if `docObject.isVisual` is `true`
3. `Std_ToggleVisibility` and other commands like it should only try to toggle visibility if `docObject.isVisual` is `true`
4. `DocumentObject` could have `true` as default
5. All classes which inherits from `DocumentObject` that isn't represented in the 3d view should set`isVisual` to `false`

I also noticed that the logic for displaying the visibility icon and handling the clicks are done manually through the feature icon. I believe we could refactor this to use toggle buttons with custom icons instead (so it looks the same).

What do you people think?
Do we need to rush this and do a quick hack like the PR above?
Or should I start with the longterm solution outlined above above instead?

FlachyJoe, 2024-09-25:
FYI, with [the filter I suggested ](https://github.com/FreeCAD/FreeCAD/pull/16817#pullrequestreview-2329368998)
[image]

Analysis is still switchable because FEM defines an _Analysis_ display mode (what that may mean?).

hyarion, 2024-09-26:
> FYI, with [the filter I suggested ](https://github.com/FreeCAD/FreeCAD/pull/16817#pullrequestreview-2329368998)
> 
> [image]

Interesting, but the techdraw template can be toggled and should have a visibility icon, right?

FlachyJoe, 2024-09-26:
> but the techdraw template can be toggled and should have a visibility icon, right?

Yes, some objects will certainly need a small fix to follow the standard. But it's only a property to set and, above all, it is their responsibility.

maxwxyz, 2024-09-26:
> Interesting, but the techdraw template can be toggled and should have a visibility icon, right?

The Material looks weird. It appears to be inside the Analysis container (which would e correct) but the indentation should be one level to the right, like the mesh. Right, @FEA-eng ?

FEA-eng, 2024-09-26:
> The Material looks weird. It appears to be inside the Analysis container (which would e correct) but the indentation should be one level to the right, like the mesh. Right, @FEA-eng ?

Yes, indentation in FEM should stay as it is

[image]

Mesh tools are further indented under Mesh.

hyarion, 2024-09-26:
the problem is that we need to replace the eye-icon with a blank space, otherwise it looks like the indentation and structure of the tree is broken

pieterhijma, 2025-03-28:
> Ok, this quickly spiraled away in scope!
> 
> To solve this in a good way I think we need to:
> 
> 1. Make `DocumentObject` keep a boolean/property/state which declares if it can be hidden/shown (`isVisualObject` <-- placeholder name).
> 
> 2. The eye icon and context menu visibility options should only be shown if `docObject.isVisual` is `true`
> 
> 3. `Std_ToggleVisibility` and other commands like it should only try to toggle visibility if `docObject.isVisual` is `true`
> 
> 4. `DocumentObject` could have `true` as default
> 
> 5. All classes which inherits from `DocumentObject` that isn't represented in the 3d view should set`isVisual` to `false`
> 
> What do you people think? Do we need to rush this and do a quick hack like the PR above? Or should I start with the longterm solution outlined above above instead?

I'm currently looking at this. I think we can infer 1 from the defined display modes. A VarSet and a TechDraw page have no DisplayModes (because they don't need to be rendered). Spreadsheet does have a display mode but it isn't registered, so I think this is simply wrong and can be removed from the list of display modes.

For 2. above, it should only be shown if the object has display modes.

For 3. above, it should only try to toggle if there are display modes

For 4. and 5. above, not necessary to set anymore.

pieterhijma, 2025-03-28:
Oh right, @FlachyJoe also proposes this as a filter.

pieterhijma, 2025-04-01:
I've just created PR #20573 that differentiates on the basis of display mode for the visibility icon, but I don't find it satisfying because TechDraw pages don't have a displaymode (rightfully so) and still it is useful to have a visibility icon.

There are two options:
- Use #20573 and add a displaymode to TechDraw objects. The drawback is that the authors of TechDraw are highly conscious of the role of the DisplayMode (see this comment by @WandererFan: `// Do not show in property editor   why? wf  WF: because DisplayMode applies only to coin and we don't use coin.`) So, to me this feels like a hack: misusing DisplayModes to hide a visibility icon.
- Create something else to differentiate document objects. This aligns with [the comment](https://github.com/FreeCAD/FreeCAD/issues/16816#issuecomment-2375282877) above from @hyarion.

However, instead of the proposal of @hyarion, I would like to propose to add a virtual function `canToggleVisibility()` to `ViewProvider.h` that returns true by default and can be set to false by the VarSet, Spreadsheet, and FEM objects. I believe that this is a better option than storing properties in the document because I consider visibility toggling as something static, part of the class and not something that should be changed by the user to suddenly make it possible to have visibility icons.

hyarion, 2025-04-01:
What I meant to say back then was that we need something new to see if we should be able to hide the document object. A virtual function on the view provider does sound like a good choice 👍

pieterhijma, 2025-04-05:
By the way, @FlachyJoe prefers the DisplayMode route as mentioned in [this comment](https://github.com/FreeCAD/FreeCAD/pull/20573#issuecomment-2769458780). Since @hyarion, @kadet1090, and I prefer the viewprovider route, I created #20649 as alternative implementation.

closed 2025-04-14 by chennes