FreeCAD/FreeCAD #16818: When click on an array in the 3D window, the base object is selected

kaiwas, 2024-09-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If we click on an array in the 3D window, the object that is the basis for the array is selected. This object is hidden.

If we make another array from the first one, the situation does not change. That is, regardless of the number of iterations, the first object is selected.

The array can be selected either in the tree view or by double-clicking in the 3D window.

Video demonstration.

https://github.com/user-attachments/assets/4452a824-f237-406a-99da-8a7b3ddc063c

In general, you can get used to this behavior, but! This method of selection does not agree with, for example, the mirror tool. I think it would be better to have the same method of selection everywhere. Uniformity is required. Because different methods of selection can lead to non-obvious results.
Earlier, I created a topic about the incomprehensible behavior of the array + mirror and only now I realized what the matter is.
https://github.com/FreeCAD/FreeCAD/issues/16751

Perhaps the selection of the "lower" object should be done using the -modifier key. For example, in the Corel Draw program, you can select an object within a group by holding down the CTRL and the left mouse button. But if I just click on the group, it is selected as a whole.

https://github.com/user-attachments/assets/745ee299-3e6b-4f54-8e69-0bfaf7a65222

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git) AppImage
Build type: Release
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * PitchedRoof
  * lattice2 1.0.0
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-09-26:
I can't reproduce the selection problem.
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38830 (Git)
Build type: Release
Branch: main
Hash: b38c42262f58cf0587b627e7a866d3756da34dc4
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```

kaiwas, 2024-09-26:
> I can't reproduce the selection problem.

I can't figure out what exactly the reason might be, but sometimes (very rarely) the array is selected normally.
I'll try to figure out the order of actions when this happens.

kaiwas, 2024-09-26:
> I can't reproduce the selection problem.

I think I figured out what's going on.
If the "link array" checkbox is checked, the base element is selected. I never paid attention to it (I had it enabled). But the strange thing is that no links are created. You can enable links only through the properties panel.

https://github.com/user-attachments/assets/d3f0c0f4-47ea-499b-bc15-93b180e9eee4

It seems that this checkbox does not work as it should.

kaiwas, 2024-09-26:
It seems that when you click on such an array, the first basic element is selected. And if the "extended array" is not active, the only element is selected.

Roy-043, 2024-09-26:
Yes, confirmed for Link arrays. This has been the case since they were implemented in V0.19.

kaiwas, 2024-09-26:
> implemented in V0.19.

The "link array" feature itself is good, because it has more capabilities than an array without this checkbox. It just needs to be "fixed" a little)

kaiwas, 2025-10-19:
relevant