FreeCAD/FreeCAD #16820: Crash when selecting x-axis on a Std_Part

hyarion, 2024-09-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:
1. Open FreeCAD
2. Create a new document
3. Create a part
4. Expand the Origin item
5. Click on X-axis

[image]

Crash!

This happens in both RC2 and current main branch (see hash in version info)

It crashes in occt because of to this call: [`src/Mod/Measure/App/Measurement.cpp`](https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Measure/App/Measurement.cpp#L353).

### Full version info

```shell
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38836 (Git)
Build type: Release
Branch: main
Hash: 6be69ac576e95ed9891b99fa6831b484be7d8679
Python 3.11.8, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CurvedShapes 1.0.10
  * sheetmetal 0.2.61
  * Curves 0.6.6
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

```
Stop reason: EXC_BAD_ACCESS (code=1, address=0x60)

* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x60)
  * frame #0: 0x000000013a25a748 libTKBRep.7.7.2.dylib`BRep_Tool::Curve(TopoDS_Edge const&, TopLoc_Location&, double&, double&) + 64
    frame #1: 0x000000013a26a3b4 libTKBRep.7.7.2.dylib`BRepAdaptor_Curve::Initialize(TopoDS_Edge const&) + 188
    frame #2: 0x000000013a26a288 libTKBRep.7.7.2.dylib`BRepAdaptor_Curve::BRepAdaptor_Curve(TopoDS_Edge const&) + 160
    frame #3: 0x000000012f9ab7ac Measure.so`Measure::Measurement::length(this=0x000000012ef6d410) const at Measurement.cpp:353:35
    frame #4: 0x000000013e21ccec MeasureGui.so`MeasureGui::QuickMeasure::printResult(this=0x000000012ef6ce80) at QuickMeasure.cpp:208:42
    frame #5: 0x000000013e21c12c MeasureGui.so`MeasureGui::QuickMeasure::tryMeasureSelection(this=0x000000012ef6ce80) at QuickMeasure.cpp:101:5
    frame #6: 0x000000013e21bd7c MeasureGui.so`MeasureGui::QuickMeasure::processSelection(this=0x000000012ef6ce80) at QuickMeasure.cpp:81:13
    frame #7: 0x000000013e21ecd0 MeasureGui.so`QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (MeasureGui::QuickMeasure::*)()>::call(f=(MeasureGui.so`MeasureGui::QuickMeasure::processSelection() at QuickMeasure.cpp:77), o=0x000000012ef6ce80, arg=0x000000016fdfbe38) at qobjectdefs_impl.h:152:13
    frame #8: 0x000000013e21ec00 MeasureGui.so`void QtPrivate::FunctionPointer<void (MeasureGui::QuickMeasure::*)()>::call<QtPrivate::List<>, void>(f=(MeasureGui.so`MeasureGui::QuickMeasure::processSelection() at QuickMeasure.cpp:77), o=0x000000012ef6ce80, arg=0x000000016fdfbe38) at qobjectdefs_impl.h:185:13
    frame #9: 0x000000013e21eacc MeasureGui.so`QtPrivate::QSlotObject<void (MeasureGui::QuickMeasure::*)(), QtPrivate::List<>, void>::impl(which=1, this_=0x000000012ef6e3c0, r=0x000000012ef6ce80, a=0x000000016fdfbe38, ret=0x0000000000000000) at qobjectdefs_impl.h:418:17
    frame #10: 0x0000000109c05dfc libQt5Core.5.15.8.dylib`___lldb_unnamed_symbol16968 + 820
    frame #11: 0x0000000109c0c96c libQt5Core.5.15.8.dylib`QTimer::timeout(QTimer::QPrivateSignal) + 52
    frame #12: 0x0000000109c01bf8 libQt5Core.5.15.8.dylib`QObject::event(QEvent*) + 104
    frame #13: 0x00000001042e4974 libQt5Widgets.5.15.8.dylib`QApplicationPrivate::notify_helper(QObject*, QEvent*) + 232
    frame #14: 0x00000001042e57ac libQt5Widgets.5.15.8.dylib`QApplication::notify(QObject*, QEvent*) + 472
    frame #15: 0x00000001053d73b0 libFreeCADGui.dylib`Gui::GUIApplication::notify(this=0x000000016fdfe2a8, receiver=0x000000012ef6e320, event=0x000000016fdfc420) at GuiApplication.cpp:83:34
    frame #16: 0x0000000109be8c74 libQt5Core.5.15.8.dylib`QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
    frame #17: 0x0000000109c24160 libQt5Core.5.15.8.dylib`QTimerInfoList::activateTimers() + 520
    frame #18: 0x000000010c02f11c libqcocoa.dylib`___lldb_unnamed_symbol3667 + 20
    frame #19: 0x000000010c02f0f8 libqcocoa.dylib`___lldb_unnamed_symbol3666 + 16
    frame #20: 0x000000018a4ae4d8 CoreFoundation`__CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 28
    frame #21: 0x000000018a4ae46c CoreFoundation`__CFRunLoopDoSource0 + 176
    frame #22: 0x000000018a4ae1dc CoreFoundation`__CFRunLoopDoSources0 + 244
    frame #23: 0x000000018a4acdc8 CoreFoundation`__CFRunLoopRun + 828
    frame #24: 0x000000018a4ac434 CoreFoundation`CFRunLoopRunSpecific + 608
    frame #25: 0x0000000194c5019c HIToolbox`RunCurrentEventLoopInMode + 292
    frame #26: 0x0000000194c4ffd8 HIToolbox`ReceiveNextEventCommon + 648
    frame #27: 0x0000000194c4fd30 HIToolbox`_BlockUntilNextEventMatchingListInModeWithFilter + 76
    frame #28: 0x000000018dd0bd68 AppKit`_DPSNextEvent + 660
    frame #29: 0x000000018e501808 AppKit`-[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 700
    frame #30: 0x000000018dcff09c AppKit`-[NSApplication run] + 476
    frame #31: 0x000000010c02f758 libqcocoa.dylib`___lldb_unnamed_symbol3677 + 968
    frame #32: 0x0000000109be6b88 libQt5Core.5.15.8.dylib`QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 312
    frame #33: 0x0000000109be9158 libQt5Core.5.15.8.dylib`QCoreApplication::exec() + 120
    frame #34: 0x00000001051d31b8 libFreeCADGui.dylib`(anonymous namespace)::tryRunEventLoop(mainApp=0x000000016fdfe2a8) at Application.cpp:2207:5
    frame #35: 0x00000001051c1568 libFreeCADGui.dylib`(anonymous namespace)::runEventLoop(mainApp=0x000000016fdfe2a8) at Application.cpp:2227:9
    frame #36: 0x00000001051c0f08 libFreeCADGui.dylib`Gui::Application::runApplication() at Application.cpp:2304:5
    frame #37: 0x000000010000c528 FreeCAD`main(argc=3, argv=0x000000016fdfeee8) at MainGui.cpp:295:13
    frame #38: 0x000000018a0460e0 dyld`start + 2360
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

AIRCAP, 2024-09-25:
Probably same issue as #16794 

Does #16799 fix this?

hyarion, 2024-09-25:
@AIRCAP It did not fix this issue. FreeCAD still crashes using the steps in the description above.

AIRCAP, 2024-09-25:
Confirmed, #16799 doesn't fix it.  Even though the cause of the crash is the exact same one as in #16794:

```
#5  0x00007fffc73caf6c in Measure::Measurement::length (this=0x555557217b40)
    at ~/src/FreeCAD/src/Mod/Measure/App/Measurement.cpp:358
358	                BRepAdaptor_Curve curve(edge);
(gdb) print edge
$1 = (const TopoDS_Edge &) @0x7fffffffc630: {<TopoDS_Shape> = {myTShape = {entity = 0x0}, myLocation = {myItems = {
        myNode = {entity = 0x0}}}, myOrient = TopAbs_EXTERNAL}, <No data fields>}

```

the edge which has its length changed question has an empty/invalid shape object

which is because the item intended to be checked is not a proper "EDGE" (its not bound by vertices)

```
(gdb) print shape
$2 = {myTShape = {entity = 0x0}, myLocation = {myItems = {myNode = {entity = 0x0}}}, myOrient = TopAbs_EXTERNAL}
(gdb) print *obj
$3 = (App::DocumentObject * const&) @0x7fffe400a1e0: 0x555558a45410
(gdb) print *subEl
$4 = "Origin.Y_Axis."
```
(I clicked on the Y axis - but it happens with any axis)

```
(gdb) print measureType
$5 = Measure::MeasureType::Line
```

which was identified as a "line" - thus triggering this code -- which it is mathematically, although an infitely long one...

so apparently `MeasureType Measurement::findType()` identifies infinite axis as "lines" - even though the actual measurement code cannot cast them as edge curves - which triggers the crash...

I have a feeling this Measure code - especially QuickMeasure - should have been much more thoroughly tested before getting merged...

AIRCAP, 2024-09-25:
code in MeasureType Measurement::findType()

```
TopoDS_Shape refSubShape;
            try {
                refSubShape = Part::Feature::getShape(*obj, (*subEl).c_str(), true);
                if (refSubShape.IsNull()) {
                    return MeasureType::Invalid;
                }
            }
            catch (Standard_Failure& e) {
                std::stringstream errorMsg;

                errorMsg << "Measurement - getType - " << e.GetMessageString() << std::endl;
                throw Base::CADKernelError(e.GetMessageString());
            }

            switch (refSubShape.ShapeType()) {

```

gives

```
(gdb) print refSubShape.ShapeType()
$6 = TopAbs_EDGE
(gdb) print (*subEl)
$
7 = "Origin.X_Axis."
```

and then

```
                case TopAbs_EDGE: {
                    edges++;
                    TopoDS_Edge edge = TopoDS::Edge(refSubShape);
                    BRepAdaptor_Curve sf(edge);

                    if (sf.GetType() == GeomAbs_Line) {
                        lines++;
                    }
```

```
(gdb) print sf.GetType()
$9 = GeomAbs_Line
```

AIRCAP, 2024-09-25:
I don't exactly get why *here* the getShape returns Edge -- but the Measurement::getShape tries to return the root shape - which ends up a null shape -  and this causes a segfault -- anyway not returning null shapes and throwing an exeption instead avoids this crash and any similar ones like it.  I'm making a PR

PR #16823 (2024-09-25): Measurement: Avoid null shapes to be returned in all code branches
 (Fix  #16820)

Returning a null shape would have triggered an exception in the bottom branch but not in the top one - which later leads to a segfault - solution - raise an exception. OOC doesn't like null shapes.
commit 69676a0f (2024-09-25): Measurement: Avoid null shapes to be returned in all code branches (Fix #16820)

Returning a null shape would have triggered an exception in the bottom branch but not in the top one - which later leads to a segfault - solution - raise an exception. OOC doesn't like null shapes.
commit d4db6e14 (2024-09-25): Measurement: Avoid null shapes to be returned in all code branches (Fix #16820)

Returning a null shape would have triggered an exception in the bottom branch but not in the top one - which later leads to a segfault - solution - raise an exception. OOC doesn't like null shapes.
closed 2024-09-25 by chennes