FreeCAD/FreeCAD #16824: view rotation in trackpad mode continues after rotation key released

inieves, 2024-09-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In FC 0.21.2 on MacOS 14.7
In Touchpad mode

Expected Behavior:
After "option" key is released after rotating view, moving cursor does not rotate the view.

Actual Behavior:
After "option" key is released after rotating view, moving cursor does rotate the view slightly.

### Full version info

```shell
[code]
OS: macOS 14.7
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * A2plus 0.4.64a
  * fasteners 0.5.20
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-25:
Please test on RC2 (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/1.0rc2)

hyarion, 2024-09-25:
I can confirm it on RC2 on mac.
<details><summary>Version info: 1.0.0RC2.38806 - macos14.5 </summary>
<p>
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CurvedShapes 1.0.10
  * sheetmetal 0.4.26
  * Curves 0.6.6

</p>
</details> 

https://github.com/user-attachments/assets/08fb1ea9-2b4d-419d-b45c-f76c514f5a6a

In this screen recording above, you will see the following:
1. the mouse is entering the view in the bottom right corner
2. option (alt on a non-mac) is pressed and not released
3. the mouse is moved and the view starts to rotate, the cursor also changes to a rotate cursor
4. the mouse stops moving and the rotation stops as well
5. option (alt) is now released, but the red dot indicating the rotation center is still on screen
6. the mouse is quickly moved to the left. The view is rotated one final step.
7. video is over.

What I think is happening is that the rotation mode is not deactivated when the modifier key is released. Instead it deactivates after the mouse event. This would explain why the red dot is showing when key is released.

The red dot doesn't show before you start moving the cursor either, which looks like the same issue is there too.

---

While this clearly is a bug, the impact is minimal. I hadn't noticed it before.

A workaround is to give FreeCAD another mouse event, like a pressing a mouse button (or clicking on the track pad). That will abort the rotation state so the view doesn't jump when mouse is moved afterwards. 

@inieves Is this the behaviour you see as well?

Rexbas, 2024-09-25:
I can reproduce this. I am already looking at the Touchpad nav style for another small problem. I will try to look for a fix for this too.

inieves, 2024-09-25:
Yes this is the behavior I see.

Rexbas, 2024-10-05:
It looks like `View3DInventorViewer::processSoEvent(const SoEvent* ev)` is never called when the event is a `SoKeyboardEvent` triggered by the left alt key. So when it is pressed, it won't immediately go into dragging mode and when it is released it won't go to idle mode. Only after another key or mouse event was triggered it syncs the state of the alt button but this may already be too late which is why we see the view move.

I think the menu bar already processes the alt key and then it does not propagate anymore?! Not sure how to fix this.

hyarion, 2024-10-05:
I tried setting a breakpoint on `src/Gui/Quarter/Keyboard.cpp` line 75: `return PRIVATE(this)->keyEvent((QKeyEvent *) event);`
The program stops on break point for Shift, Command, and Ctrl (this is on macOS), but not on option(alt).

This method is responsible to convert all qt keyboard events to coin counterparts. As the event hasn't reached coin yet, it can't be coin related. It must be Qt related as @Rexbas mentions above.

luzpaz, 2025-11-11:
@inieves can you test on development build ? https://github.com/FreeCAD/FreeCAD/releases

inieves, 2025-11-15:
> luzpaz
>  left a comment
> (FreeCAD/FreeCAD#16824)
> i cannot test 2025.11.12, macos download not available
>
> @inieves <https://github.com/inieves> it is now. https://github.com/FreeCAD/FreeCAD/releases click the 'Assets' to expand the subsection.

i verified 2025.11.12 does not pass