FreeCAD/FreeCAD #16827: BIM: When choosing the proxy work plane in bim, all levels are turned on

gianlucafiore, 2024-09-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]
I am working on a building with its independent structure, which I am now drawing the Architecture for. When I want to click on a work plane, previously turning off all the levels, the floor plan I want is selected correctly, but all the other floors are turned on and it takes me to a **perspective floor view** at level 0.

[image]
The annoying thing about this is that I have to navigate to the floor I selected, put the view back into orthogonal mode, and turn off all the floors that I don't need to manipulate or that block my view of the floor plan.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Spanish/Argentina (es_AR)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Qt default
Installed mods: 
  * Dracula 0.0.5
  * OpenDark 2024.3.13
  * OpenTheme 2024.9.1
  * SearchBar 1.0.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-09-26:
This is intended behavior. It depends on 2 properties.

Please read up on those properties in the Wiki:
Restore State
Restore View
https://wiki.freecad.org/Draft_WorkingPlaneProxy#Properties

Closing this.

closed 2024-09-26 by Roy-043

yorikvanhavre, 2024-10-02:
Aren't you simply not using the "Restore view" and "Restore state" properties?

[image]

If you set those two to False, no view change happens when you double-click a WP.

Note that you can also write a new camera view or objects state for each one... Just right-click the WP and select "write camera" or "write objects state"

yorikvanhavre, 2024-10-02:
*EDIT* oops weirdly I didn't see @Roy-043 's answer, sorry