FreeCAD/FreeCAD #16828: Crash on opening old project

goodman-capu, 2024-09-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Old project created with 0.21.2, now using 1.0.0 RC2
It will crash if I double click file to open without FreeCAD already opened
It will **not** crash if I open FreeCAD first and then open the project

Project file:
[Blind Roller.FCStd.zip](https://github.com/user-attachments/files/17137443/Blind.Roller.FCStd.zip)

A similar [issue](https://github.com/FreeCAD/FreeCAD/issues/15778) but that was supposed to be fixed in RC2? So maybe separate problem.

### Full version info

```shell
OS: macOS 15.0
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hyarion, 2024-09-25:
I've tested the file and it crashes 1.0RC2 for me too.

OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CurvedShapes 1.0.10
  * sheetmetal 0.4.26
  * Curves 0.6.6

It's strange that not only does it load the file without crashing when loaded from gui, but also when starting freecad with the file from the command line `/Applications/FreeCAD.app/Contents/MacOS/FreeCAD '/Users/nauck/Downloads/Blind Roller.FCStd'`

JohnOCFII, 2024-09-25:
I was able to successfully double-click on the file above and it opened successfully for me. I'm assuming this was the unadulterated 0.21.2 file), not one that had been opened and saved.

Note that this was one of the post-RC releases.

Un-related, this did trigger the dark text issue which seems to happen when the theme (in this case OpenDark) differs from the OS display setting) in this case, `Light Mode.`

```
OS: macOS 15.0
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38827 (Git)
Build type: Release
Branch: main
Hash: abd5c3ca31539f8ed37ac3dec2b9abd252a5310d
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * Curves 0.6.47
```

[image]

goodman-capu, 2024-09-26:
@JohnOCFII This project file was indeed opened in 1.0.0RC2 before and applied the automatic "fixes", but I don't think it made any difference. I have a backup project file that is indeed "unadulterated 0.21.2", and it also crashed.

hyarion, 2024-09-26:
It doesn't seem to be related to just old files.

I've found that the following steps also results in a crash:
1. Open FreeCAD RC2
2. Create new document
3. Save the new document
4. Close FreeCAD
5. Double click on the file
6. Splash screen starts to show for a split second then a crash. The main window never opens.

Here's the file I saved:
[Unnamed.FCStd.zip](https://github.com/user-attachments/files/17144636/Unnamed.FCStd.zip)

My full version can be seen in previous post above.

The file was saved under `/Users/nauck/Downloads/Unnamed.FCStd` so there was no strange characters in the name.

<details><summary>Stack trace</summary>
<p>
Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Codes:       0x0000000000000001, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [99733]

0   libsystem_pthread.dylib       	       0x18a3c94f8 pthread_mutex_lock + 12
1   libCoin.80.0.3.dylib          	       0x10842bd58 SoGLCacheContextElement::getUniqueCacheContext() + 28
2   libFreeCADGui.dylib           	       0x105be05b0 SIM::Coin3D::Quarter::QuarterWidgetP::findCacheContext(SIM::Coin3D::Quarter::QuarterWidget*, QOpenGLWidget const*) + 184
3   libFreeCADGui.dylib           	       0x105be04ac SIM::Coin3D::Quarter::QuarterWidgetP::QuarterWidgetP(SIM::Coin3D::Quarter::QuarterWidget*, QOpenGLWidget const*) + 156
4   libFreeCADGui.dylib           	       0x105bdd078 SIM::Coin3D::Quarter::QuarterWidget::constructor(QSurfaceFormat const&, QOpenGLWidget const*) + 212
5   libFreeCADGui.dylib           	       0x105bdd3c0 SIM::Coin3D::Quarter::QuarterWidget::QuarterWidget(QWidget*, QOpenGLWidget const*, QFlags<Qt::WindowType>) + 92
6   libFreeCADGui.dylib           	       0x105be3b08 SIM::Coin3D::Quarter::SoQTQuarterAdaptor::SoQTQuarterAdaptor(QWidget*, QOpenGLWidget const*, QFlags<Qt::WindowType>) + 52
7   libFreeCADGui.dylib           	       0x105c0c390 Gui::View3DInventorViewer::View3DInventorViewer(QWidget*, QOpenGLWidget const*) + 44
8   libFreeCADGui.dylib           	       0x105c07610 Gui::View3DInventor::View3DInventor(Gui::Document*, QWidget*, QOpenGLWidget const*, QFlags<Qt::WindowType>) + 252
9   libFreeCADGui.dylib           	       0x105811858 Gui::Document::createView(Base::Type const&) + 256
10  libFreeCADGui.dylib           	       0x10578e0c4 Gui::Application::slotNewDocument(App::Document const&, bool) + 2540
11  libFreeCADApp.dylib           	       0x1044b37b0 boost::signals2::detail::slot_call_iterator_t<boost::signals2::detail::variadic_slot_invoker<boost::signals2::detail::void_type, App::Document const&, bool>, std::__1::__list_iterator<boost::shared_ptr<boost::signals2::detail::connection_body<std::__1::pair<boost::signals2::detail::slot_meta_group, boost::optional<int>>, boost::signals2::slot<void (App::Document const&, bool), boost::function<void (App::Document const&, bool)>>, boost::signals2::mutex>>, void*>, boost::signals2::detail::connection_body<std::__1::pair<boost::signals2::detail::slot_meta_group, boost::optional<int>>, boost::signals2::slot<void (App::Document const&, bool), boost::function<void (App::Document const&, bool)>>, boost::signals2::mutex>>::dereference() const + 96
12  libFreeCADApp.dylib           	       0x1044b31dc boost::signals2::detail::signal_impl<void (App::Document const&, bool), boost::signals2::optional_last_value<void>, int, std::__1::less<int>, boost::function<void (App::Document const&, bool)>, boost::function<void (boost::signals2::connection const&, App::Document const&, bool)>, boost::signals2::mutex>::operator()(App::Document const&, bool) + 600
13  libFreeCADApp.dylib           	       0x104697278 App::Application::newDocument(char const*, char const*, bool, bool) + 8412
14  libFreeCADApp.dylib           	       0x10469cf0c App::Application::openDocumentPrivate(char const*, char const*, char const*, bool, bool, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>>&&) + 532
15  libFreeCADApp.dylib           	       0x1046cba9c App::Application::openDocuments(std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>> const&, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>> const*, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>> const*, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>>*, bool) (.504) + 1976
16  libFreeCADApp.dylib           	       0x104698fd4 App::Application::openDocument(char const*, bool) + 204
17  libFreeCADApp.dylib           	       0x1046d1998 App::Application::sOpenDocument(_object*, _object*, _object*) + 312
18  libpython3.11.dylib           	       0x103967e20 cfunction_call + 124
[... cut ...]
48  libFreeCADGui.dylib           	       0x105890d00 Gui::StartupPostProcess::execute() + 40
49  libFreeCADGui.dylib           	       0x105799430 Gui::Application::runApplication() + 1492
50  freecad                       	       0x102ed7248 main + 3464
51  dyld                          	       0x18a0460e0 start + 2360
</p>
</details>

Just as previously the file can be loaded fine from both the menu and from command line `/Applications/FreeCAD.app/Contents/MacOS/FreeCAD /Users/nauck/Downloads/Unnamed.FCStd`

luzpaz, 2024-09-26:
This is feeling like 'Blocker' territory.

hyarion, 2024-09-26:
I agree that it must be macos related.

It seems like it is crashing here in coin: https://github.com/coin3d/coin/blob/2e7e4d041a9bce29ac288ab1aa40d4ed5a00bac7/src/elements/GL/SoGLCacheContextElement.cpp#L539C17-L539C30

Which probably means that the mutex hasn't been fully created at that point.

Did anything change in how documents are loaded lately, like moved from being handled with signals/slots to stuff called directly in a function? (that could change the order of some operations)

JohnOCFII, 2024-09-26:
> Did anything change in how documents are loaded lately,

Could it have anything to do with the recently added pop-up regarding the need to re-compute older files?

hyarion, 2024-09-26:
> > Did anything change in how documents are loaded lately,
> 
> Could it have anything to do with the recently added pop-up regarding the need to re-compute older files?

Oh right! A related change actually did modify a code path from being handled using signals to just being called in order in the openDocument function! Thanks, I'll see if I can change that back temporarily to make this go away.

hyarion, 2024-09-27:
This is the commit that I suspect could be causing the issue: https://github.com/FreeCAD/FreeCAD/commit/f82bfd5ddcddfa96359789093190c639ab088e24

But since it is a bug that only happens when double clicking a freecad document I'm having trouble running it with my dev builds.

maxwxyz, 2024-09-27:
@chennes FYI

bgbsww, 2024-09-27:
Well to be clear, there was no code path, then there was the one using signals which caused other breakage, and was simplified to the one that just runs after the document opens.

There are three places that call the recompute check.  Two of them are in the StartView that you presumably never see in this scenario, so this would have to be the one in Application::open  if you comment out that checkForRecomputes does the problem go away?

hyarion, 2024-09-27:
@bgbsww This only happens on macos if FreeCAD is built as a self contained application bundle and opened with a document.

I cannot debug this as I cannot build FreeCAD as a self contained application bundle on my system.

Do we have any other macos developer who we can ping?

I have a stack trace in this comment: https://github.com/FreeCAD/FreeCAD/issues/16828#issuecomment-2376279903
It shows it crashing in `1 libCoin.80.0.3.dylib 0x10842bd58 SoGLCacheContextElement::getUniqueCacheContext() + 28
` when accessing a mutex. That lets me believe it calls something in the wrong order now.

bgbsww, 2024-09-27:
Wish I had a more modern Mac here - last one is still an Intel.  Somebody send me one!

In any case, since we can't prove the point I'll just go for it:  This is a splash screen related issue which is why it didn't show up in RC1.  If it were a recompute prompt issue it wouldn't happen with a new file.

Should I be proven wrong, and that happens all the time, I'll happily dig for something broken on the recompute prompt path.

JohnOCFII, 2024-09-28:
> Do we have any other macos developer who we can ping?

@oursland is a macOS FreeCAD developer 

I can build on macOS arm, but I've not built the app bundle -- just the standard Conda build as described in the Developers Handbook: https://freecad.github.io/DevelopersHandbook/gettingstarted/

hyarion, 2024-09-28:
> Wish I had a more modern Mac here - last one is still an Intel. Somebody send me one!
>
> In any case, since we can't prove the point I'll just go for it: This is a splash screen related issue which is [...]

I verified that RC1 works for me, and it does.
You are probably right (and I'm probably wrong), only testing will tell.

The next step is for someone to manage to reproduce this with their own builds (by building bundles) and then bisect between RC1 and RC2.

hyarion, 2024-10-03:
> In any case, since we can't prove the point I'll just go for it: This is a splash screen related issue which is why it didn't show up in RC1. If it were a recompute prompt issue it wouldn't happen [...]

It seems like @bgbsww was correct 👍

hyarion, 2024-10-03:
However, splash screen is shown _over_ the recompute dialog.

hyarion, 2024-10-09:
Tried with a new weekly build, still same issue with splash screen over dialog.

@maxwxyz you can remove the label "Crash", but it is still a "Blocker" in my opinion.

@oursland is this issue something you can look into?

luzpaz, 2024-10-09:
Any proposal for renaming this ticket ?

goodman-capu, 2024-10-09:
> Any proposal for renaming this ticket ?

I renamed it to better reflect the true problem

hyarion, 2024-10-09:
@goodman-capu This was a crash, now it isn't.

I think it can be renamed to:
> Unexpected behavior when opening Open FreeCAD.app with older documents.

I think that both describes the previous crash and the current issue with hidden dialog under splash screen that never goes away.

goodman-capu, 2024-10-09:
> @goodman-capu This was a crash, now it isn't.
> 
> I think it can be renamed to:
> 
> > Unexpected behavior when opening Open FreeCAD.app with older documents.
> 
> I think that both describes the previous crash and the current issue with hidden dialog under splash screen that never goes away.

This issue is about the crash in RC2, so I think it accurately describes it? If it's fixed in some future weekly builds then we can mark this bug as fixed.

About the "hidden dialog under splash screen", I don't experience it and don't know what it means. But feel free to start a new issue for that :)

oursland, 2024-10-09:
@hyarion 

IIRC, I confirmed the crash in 1.0.0RC2 and that it was fixed in a recent weekly build.  From that perspective, this issue should be closed as fixed.

The issue of the recompute dialog being hidden should be a separate issue.

I don't have much time right now to investigate and resolve the hidden dialog issue.  It seems like it should be a matter of calling `getMainWindow()->stopSplasher()` before the call to [`QMessageBox::warning()` in `src/Gui/Application.cpp`](https://github.com/FreeCAD/FreeCAD/blob/main/src/Gui/Application.cpp#L971).  Note that recent changes to `stopSplasher()` have introduced a 3 second delay to destroying the splash screen, which may not be appropriate when attempting to get user confirmation.  There may be other modal dialogs triggered during startup which should receive this treatment, which may merit either a generalized approach to modal creation or triggering the `stopSplasher()` in some other way.

closed 2024-10-10 by maxwxyz