FreeCAD/FreeCAD #16829: Measure tool: Always stays on

gianlucafiore, 2024-09-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When making measurements, and pressing ESC impulsively to close the command, the command closes correctly, but the information balloons of the measurements are still visible, bothering the sight, when you invoke the command again, you can make new measurements, deleting the previous ones that were bothering, and if you press the close button they disappear correctly, but in case of pressing ESC again the Bug remains that they are still visible

[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Spanish/Argentina (es_AR)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Qt default
Installed mods: 
  * Dracula 0.0.5
  * OpenDark 2024.3.13
  * OpenTheme 2024.9.1
  * SearchBar 1.0.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

masterxq, 2024-09-28:
Not sure if this is the same bug, but I can confirm that there are multiple problems with this.
One way to reproduce one of the problems is:
1. Open the measurement tool
2. make a measurement
3. Press reset
4. Close the view
5. Messurement will stay and not appear in the tree view

[image]

[image]

```
OS: Debian GNU/Linux trixie/sid (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * fcgear 1.0.0
  * kicadStepUpMod 10.18.6
  * Curves 0.6.8
  * Dracula 0.0.5
```

There are really many bugs in the measurement tool, it should be tested by a developer and all problems should be solved ASAP, you will create a lot of management overheads with all the bug tickets that will be created if this is not cleaned up ASAP.

luzpaz, 2024-09-29:
Can you please test RC2

pixelwaster, 2024-09-29:
It is happening in RC2. The reset button only clears the window in the task pane but leaves the balloons. The last measurement is redrawn when restarting the tool, 

OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * AddonManager
  * Assembly
  * BIM
  * btl 0.9.9
  * CAM
  * CurvedShapes 1.0.10
  * Curves 0.6.47
  * Draft
  * FeedsAndSpeeds 0.5.0
  * Fem
  * Help 1.0.3
  * Idf
  * Import
  * Inspection
  * lattice2 1.0.0
  * Manipulator 1.5.7
  * Material
  * Measure
  * Mesh
  * MeshPart
  * OpenSCAD
  * Part
  * PartDesign
  * Plot 2022.4.17
  * Points
  * ReverseEngineering
  * Robot
  * sheetmetal 0.4.26
  * Show
  * Silk 0.1.5
  * Sketcher
  * Spreadsheet
  * Start
  * Surface
  * TechDraw
  * Test
  * Tux
  * Web

luzpaz, 2024-09-29:
Thanks @pixelwaster (my bad though since there is a PR pending that I had overseen so it makes sense that it was still broken in RC2)

closed 2024-09-30 by yorikvanhavre

gianlucafiore, 2024-10-11:
when does a new version of RC2 come out again? The problem still persists

duub, 2024-11-25:
This issue still happens in v1.0
-Reset or Close button doesn't takes any effect.
As you can see in the screenshot the Result is empty but there is still the buble with the measure.

I only could remove the measure if I make a measure, save it and then delete the item in the left menu with all the stuff in the measurements folder.

[image]