FreeCAD/FreeCAD #16830: ASSEMBLY: View -->Draw Style cannot be changed from As is --> Points AND NOT --> Wireframe

MakerFC, 2024-09-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I loaded successfully an assembly consisting of 4 parts in the new wb. 
It is not possible to change bodies Draw Style. If you select e.g. one body the View **cannot** be changed from As is
- Points
AND NOT 
- Wireframe

Changes to 
- Hidden line
- No shading
- Shaded
- Flat lines
are working as usual.

[Assembly24_Test_constraints.zip](https://github.com/user-attachments/files/17137811/Assembly24_Test_constraints.zip)

### Full version info

```shell
OS: openSUSE Leap 15.5 (KDE//usr/share/xsessions/default)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * sheetmetal 0.4.23
  * CurvedShapes 1.0.9
  * Render 2024.8.4
  * A2plus 0.4.65
  * Silk 0.1.5
  * Defeaturing 1.2.2
  * fasteners 0.5.25
  * Curves 0.6.42
  * Utils 0.2.5
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-09-26:
You need to supply the linked files of your assembly.

luzpaz, 2024-09-26:
I find the summary confusing, can someone suggest a more descriptive title ?

MakerFC, 2024-09-27:
> You need to supply the linked files of your assembly.

Here they are: :-)
[Test_Plates.zip](https://github.com/user-attachments/files/17164624/Test_Plates.zip)

Roy-043, 2024-09-28:
Thanks for the files. I can reproduce this. The assembly has to have external links for this to occur.

eduncan911, 2024-11-23:
I can confirm this is still present in 1.0.0 release.

FEA-eng, 2025-10-21:
@MakerFC I can't reproduce it. Is this how you change it?

[image]

Can you recheck in the newest weekly build?

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/10/14 15:14:51
Build type: Release
Branch: main
Hash: 41ce37233d64c81c2f105de085635d4dcd669943
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
```

luzpaz, 2026-02-28:
@MakerFC 👆

PaddleStroke, 2026-03-04:
Some progress here. It appears that it is broken only for external links to bodies. External link to a Part workbench box works fine.

PaddleStroke, 2026-03-04:
Ok so it appears that this is linked to the 'DisplayModeBody' property. If it's set to Through it doesn't work. If set to Tip then it works. Although it seems buggy.

Now the big question is: What the hell is this property. Even looking at the code I cannot understand clearly what it is. This feels like it should not exist.