FreeCAD/FreeCAD #16833: Sketch: defining a symmetry line / rotational axis for dimensioning diameters is not possible (profiles of rotational parts)

imaluschi, 2024-09-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Dimensioning a sketch for the profile / cross section of a rotational part: it is not possible to dimension diameters because it is not possible to define a symmetry line / rotational axis.

[image]

After years of working with professional CAD-Software (Autodesk Mechanical and Inventor, Unigraphics NX and Siemens PLM NX, CATIA V5) in different companies: nobody, really nobody dimensions such sketches with radii.

Doing such a sketch in CATIA V5:
After drawing a line you can define it as a symmetry line
(The german text in the screenshot: axis generation "The selected line will be converted into an axis. Should the line be converted?")
[image]

When dimensioning you can choose if you want to define the diameter instead of the radius.
[image]

The fully defined sketch.
[image]

When creating the rotation the rotational axis is recognised automatically - because it was already defined in the sketch. Of course you can choose another rotational axis (no matter whether it makes sense).
[image]

And no, defining the diameters as parameters and dimension with parameter/2 is no alternative.
For a part like this would that be impossible - at least without a table of all diameters. And if there are changes (in the part) it is unmanageable.
[image]

Of course FreeCAD is not CATIA or any other professional CAD-software - and should not be! But at least the working method should be possible.
The attempt to switch to FreeCAD failed for the moment. Maybe one attemp more if the described working method will be possible. I don't feel like spending more than twice the time is a way to go.

### Full version info

```shell
OS: Fedora Linux 40 (KDE Plasma) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38842 (Git)
Build type: Release
Branch: main
Hash: 69676a0f9d2d5b288bda3d1c6dbad5a6ec26bfaa
Python 3.12.6, Qt 6.7.2, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Austria (de_AT)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/breeze
Installed mods: 
  * sheetmetal 0.4.26
  * Defeaturing 1.2.2
  * InventorLoader 1.4.0
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-09-26:
Duplicate of #7334

Roy-043, 2024-09-26:
Closing this as a duplicate.

closed 2024-09-26 by Roy-043