FreeCAD/FreeCAD #16834: Menu - Preference > Crash

yagikjp, 2024-09-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When select [Preference] on menu bar, FreeCAD crash.

### Full version info

```shell
FreeCAD 1.0.0 Release Candidate 2
FreeCAD_1.0.0RC2-conda-macOS-x86_64-py311.dmg

on MacOS 14.6.1
Intel Core i7
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

JohnOCFII, 2024-09-25:
Thanks for creating the Github issue.

- Did you run a previous version of FreeCAD (0.21 or older)?
- What language did you have selected?

The next time you launch FreeCAD, before trying to access Preferences, please select the `About` menu (just above the `Preferences` selection) and you should see a dialog box displayed.  There is a button "Copy to Clipboard" on that window.  Please select that option, then come back to this issue and add a reply with the complete information that you can paste into the ticket.  It should include information similar to what I included below::

```
OS: macOS 15.0
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38827 (Git)
Build type: Release
Branch: main
Hash: abd5c3ca31539f8ed37ac3dec2b9abd252a5310d
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * Curves 0.6.47
```

yagikjp, 2024-09-26:
Thank you for quick response.
- Yes, I using 0.20.2 too. (build revision 29603)
- on 0.20.2 selected language is Japanese, but after switch English on 0.20.2, 1.0.0 also crash.

and About menu info is as follows.

OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
Installed mods: 
  * slic3r-tools
  * fcgear 1.0.0
  * fasteners 0.4.70

xtemp09, 2024-09-26:
@yagikjp, can you upload the crash report?

----

> * Yes, I using 0.20.2 too. (build revision 29603)

Does it mean that _0.20.2_ works as expected and the problem shows up only starting from _1.0.0_?

> * on 0.20.2 selected language is Japanese, but after switch English on 0.20.2, 1.0.0 also crash.

Does it mean that _FreeCAD 1.0.0_ crashes when you switch to English, but works fine when Japanese (system language) is used?

----

Recently the issue tracker has began to receive crash reports; too many of them are caused by macros, add-ons, some old options left from previous versions or mods.

@yagikjp, I recommend:
- reset the settings and launch _FreeCAD_. [^1]

[^1]: https://wiki.freecad.org/Frequently_asked_questions#FreeCAD_was_running_normally,_and_suddenly_it_doesn't_start_anymore

yagikjp, 2024-09-26:
Yes, when using 0.20.2, it can open preference window, but 1.0.0 can not open because FreeCAD crash.

on 1.0.0 can not change language, because before open preference window, FreeCAD is crash down.
at first, on 0.20.2 select Japanese or English, then close 0.20.2, after that open 1.0.0, both of Japanese / English, preverence window crash down.

xtemp09, 2024-09-26:
@yagikjp, how does resetting configuration files affects it? You can back up settings if you intend to keep them.

yagikjp, 2024-09-26:
how does resetting configuration files?
I don't know that operation. please let me know.
thanks.

xtemp09, 2024-09-26:
@yagikjp, 

https://wiki.freecad.org/Frequently_asked_questions#FreeCAD_was_running_normally,_and_suddenly_it_doesn't_start_anymore

yagikjp, 2024-09-26:
after delete [user.cfg], it works.
thanks!

maxwxyz, 2024-09-27:
Closing this then.

closed 2024-09-27 by maxwxyz