FreeCAD/FreeCAD #16835: When importing or opening an STL file, if the path contains Chinese characters, the import will fail.

ronggang, 2024-09-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When importing or opening an STL file, if the path contains Chinese characters, the import will fail, and the error message is as follows:
```
10:40:17  pyException: Traceback (most recent call last):
  File "D:\Tools\FreeCAD_1.0.0RC2-conda-Windows-x86_64-py311\bin\Lib\site-packages\freecad\module_io.py", line 12, in OpenInsertObject
    except PyExc_FC_AbortIOException:
           ^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'NameError'>: name 'PyExc_FC_AbortIOException' is not defined
```
This error does not occur in FC 1.0.0RC1.
---------------------------
The above information was translated by ChatGPT.
-----------------------
导入或打开stl文件时，如果路径包含中文，则会导入失败。会提示以上信息
但这个错误在 FC 1.0.0RC1 上不会出现。

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Chinese/China (zh_CN)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * Assembly4 0.50.15
  * fasteners 0.5.28
  * freecad.gears 1.2.0
  * ProDarkThemePreferencePack 1.0.0
  * Render 2024.8.4
  * sheetmetal 0.4.25
```

### Subproject(s) affected?

None

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Phakrada666, 2024-09-26:
****

hyarion, 2024-09-26:
This is a duplicate of https://github.com/FreeCAD/FreeCAD/issues/16743

Syres916, 2024-09-26:
> This seems related to #16743

Almost definitely, updates to the root cause and possible fix will be kept over in #16743

ronggang, 2024-09-26:
> > This seems related to #16743这似乎与 #16743
> 
> Almost definitely, updates to the root cause and possible fix will be kept over in #16743几乎可以肯定的是，对根本原因和可能的修复的更新将保留在 #16743

Looks like the same problem.

luzpaz, 2024-09-26:
Maybe somone can create a unit test that tests for unicode in paths ?

hyarion, 2024-09-26:
The problem is indeed unicode characters.

FreeCAD automatically replaces `\` to `/`, even for unicode characters encoded like: ´\uXXXX´.

This happens in fileinfo.cpp on line 242 to 246: https://github.com/FreeCAD/FreeCAD/blob/main/src/Base/FileInfo.cpp#L247
<details><summary>Backtrace</summary>
<p>
(I google translated "random chinese characters" and used that to test. I blame google if I accedentally wrote any profanity)

  * frame #0: 0x000000010146f078 libFreeCADBase.dylib`Base::FileInfo::setFile(this=0x000000016fdf9bf0, name="/Users/nauck/Downloads/随机汉字/pawl2.step") at FileInfo.cpp:246:31
    frame #1: 0x000000010146ef24 libFreeCADBase.dylib`Base::FileInfo::FileInfo(this=0x000000016fdf9bf0, fileName="/Users/nauck/Downloads/随机汉字/pawl2.step") at FileInfo.cpp:99:5
    frame #2: 0x000000010146f0f0 libFreeCADBase.dylib`Base::FileInfo::FileInfo(this=0x000000016fdf9bf0, fileName="/Users/nauck/Downloads/随机汉字/pawl2.step") at FileInfo.cpp:98:1
    frame #3: 0x00000001051b7774 libFreeCADGui.dylib`Gui::Application::importFrom(this=0x000000016fdfe0b8, FileName="/Users/nauck/Downloads/随机汉字/pawl2.step", DocName="Unnamed", Module="ImportGui") at Application.cpp:691:20
    frame #4: 0x0000000105b8ca30 libFreeCADGui.dylib`Gui::MainWindow::loadUrls(this=0x000000016fdfe080, doc=0x00000001385ea200, urls=0x000000016fdfa010) at MainWindow.cpp:2433:32
    frame #5: 0x0000000105b8bedc libFreeCADGui.dylib`Gui::MainWindow::dropEvent(this=0x000000016fdfe080, e=0x000000016fdfa7c0) at MainWindow.cpp:2183:9
    frame #6: 0x0000000104309f4c libQt5Widgets.5.15.8.dylib`QWidget::event(QEvent*) + 104
</p>
</details>

I'm not 100% on how to fix this without potentially breaking something else.

@luzpaz do you know anyone we could ping about this bug?

hyarion, 2024-10-03:
I've tested this again on the latest main and I can now successfully load files with Chinese characters in the path.
 
@maxwxyz we can close this issue

Verified on:
```
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38888 (Git)
Build type: Release
Branch: main
Hash: d423795e2dc32dc16f1e8c1835104deb01c9cab9
Python 3.11.8, Qt 5.15.8, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * A2plus 0.4.68
  * lattice2 1.0.0
  * CurvedShapes 1.0.10
  * sheetmetal 0.4.26
  * Curves 0.6.6
```

closed 2024-10-03 by maxwxyz

menkej, 2024-11-06:
I experienced a similar behavoir with 1.0 RC2 and also the latest nightly. The problematic path was
`/Users/joerg/Downloads/Haken für Schwerlastregal von Toom - 6119942/files/Part_Studio_1_-_Part_1-5.stl` - after copying the file to Downloads, so just `/Users/joerg/Downloads/Part_Studio_1_-_Part_1-5.stl` it worked fine... Maybe check again @maxwxyz

menkej, 2024-11-06:
> @menkej <https://github.com/menkej> we need full About
> <https://wiki.freecad.org/About> info for all versions you test on. Thanks

OS: macOS 15.0.1
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

luzpaz, 2024-11-06:
@menkej Thanks and the nightly you tested on ?

menkej, 2024-11-07:
Thanks for the hint @JohnOCFII ! I was able to stat it now and the import worked from the mentioned path above. Excellent!