FreeCAD/FreeCAD #16837: Changing properties of Involute Gear changes Face-ID of pad

m-ihmig, 2024-09-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have a design using "Involute Gear", created in "Part Design" Workbench.

To create pockets around the axis of a defined depth, I used Sketch+Pocket (see e.g. SketchPocketZ48 + PocketZ48 in the attached file) and attached the sketch ("support ") to the outside face of the previously created padded gear (PadZ48), e.g. Face482.

Seen behaviour:
When I now change a gear parameter in Z48 (edit Z48 and change "High Precision" from true to false),
the padded gear faces get a new ID (in my case changing from Face482 to Face386),
which causes the sketchs (PocketSketch*) to loose their support, leading to a Recalculate Failed error.

Expected behaviour:
Changing a parameter in an earlier created pad should keep the Face IDs, so that later attached supports are not lost.

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
[/code]
```

### Subproject(s) affected?

PartDesign

### Anything else?

[Gear1_04redesign.zip](https://github.com/user-attachments/files/17144476/Gear1_04redesign.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-09-26:
The problem does not occur in the current dev. The TNP mitigation code works fine here.
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38830 (Git)
Build type: Release
Branch: main
Hash: b38c42262f58cf0587b627e7a866d3756da34dc4
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```

closed 2024-09-26 by Roy-043

luzpaz, 2024-09-26:
@m-ihmig Install RC2 to see this is fixed (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/1.0rc2)

m-ihmig, 2024-09-26:
Thanks for the quick response! Just tried it on 1.0rc2 and it's working - great!