FreeCAD/FreeCAD #16838: sequoia seg fault

louiscypher666, 2024-09-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Both free cad pre-release versions RC[12] segfault on startup

### Full version info

```shell
Freecad 1.0.0 RC1 (and RC2)
```

### Subproject(s) affected?

None

### Anything else?

Process:               freecad [847]
Path:                  /Applications/FreeCAD.app/Contents/Resources/bin/freecad
Identifier:            freecad
Version:               ???
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-09-26 08:27:59.5550 +0100
OS Version:            macOS 15.0 (24A335)
Report Version:        12
Bridge OS Version:     9.0 (22P353)
Anonymous UUID:        392FA8BC-0A36-E3BC-C1CB-F2EF21B632EA

Time Awake Since Boot: 120 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000039
Exception Codes:       0x0000000000000001, 0x0000000000000039

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [847]

VM Region Info: 0x39 is not in any region.  Bytes before following region: 4308103111
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      100c87000-100ca7000    [  128K] r-x/r-x SM=COW  /Applications/FreeCAD.app/Contents/Resources/bin/freecad

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libQt5Core.5.15.13.dylib      	       0x105703004 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 106

Thread 1:

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

louiscypher666, 2024-09-26:
I've tried both RC versions, rebooted etc but still segfaults

louiscypher666, 2024-09-26:
Fixed by removing previous version data from ~/Library/Application Support/FreeCAD

hyarion, 2024-09-27:
From looking at the crash report it looks like something funny happened in qt, why is unknown.
There are not enough information in this report to continue the investigation.

@louiscypher666 do you know how to reproduce this issue, like did you keep the files you removed so this could be tested more or should we close the issue?

louiscypher666, 2024-09-27:
I suspect I could re-create it if I reinstalled 0.21, ran it and then re-installed rc2. I’ll give it a try

luzpaz, 2024-10-14:
@louiscypher666 any luck ?

louiscypher666, 2024-10-14:
I'm sorry fr the delay - I've been away. I will endeavour to get to this in the next few days.

louiscypher666, 2024-10-16:
I tried and unfortunately it works as expected - no crash. Sadly as I failed to retain the original files, I suspect there was something historical in there (not sure when it was added, which version did it) but re-running 0.21 then rc2 no longer core dumps.

hyarion, 2024-10-16:
I think we can close this issue as we don't have enough information and op can't reproduce the crash anymore.

It can be reopened if the same thing happens again.

closed 2024-10-16 by luzpaz