FreeCAD/FreeCAD #16839: Export filename 1.0.0RC2

linuxdudeg, 2024-09-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Select a body, click on FIle.Export and the filename is preloaded.
Now close file and open another file. 
Select body, click on File.Export and the preloaded filename is not updated, it always holds the filename from the first export

### Full version info

```shell
OS: TUXEDO OS 3 (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * package.xml
  * README.md
  * ExtremeProLight 1.6.0
  * LICENSE
  * INSTALL.md
  * resources
  * screenshot.png
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

mpictor, 2024-10-16:
Interesting, this is slightly different from what I've experienced. For me, the export name is correct (`$file_name-$shape_name`) each time, until I override. Once I do that, it gets "stuck", always suggesting whatever name I last saved with. I haven't found a way to go back to automatically choosing a name, short of exiting and re-launching freecad.

A bit annoying, as sometimes I'm happy with `filename-basename` and other times I want just `basename`.

I've seen it with 1.0.0RC2, 0.21.2, and possibly with older versions.

```
OS: Gentoo Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
Installed mods: 
  * Lithophane
  * OpticsWorkbench 1.0.9
```

luzpaz, 2024-10-16:
Can someone propose a less generic title for this ticket ?

NewStorm, 2024-10-19:
@luzpaz "Export filename not switching back to default (after opening new file)" ?

I btw can not reproduce OPs issue either. The (flawed) behavior I observed matches with that of @mpictor.
It can be solved by checking if the last document where the proposed filename was overwritten is still the active one. If not, the default kicks back in.
IMHO should the overwritten path still be the proposed one because having a dedicated export directory is a likely workflow.

I can make a PR for it.

luzpaz, 2024-10-19:
@NewStorm sounds good. Feel free to make a PR and it can go through review :pray:  
BTW, can you always add your full [About](https://wiki.freecad.org/About) info when testing issues so we can have as much info as possible about the systems issues can/can't be reproduced on. TIA

PR #17354 (2024-10-21): Export Dialog:Switch back to default filename 
Currently, if the user chooses to overwrite the proposed filename in the export dialog, the new filename becomes the new default even after opening a different file.
This PR fixes that by detecting that the active file has been changed, and then suggesting the default again.
Fixes #16839
commit 6f87fc09 (2024-10-21): Switch back to default filename for exports after the active document changes. Fixes #16839
commit bad2d2a3 (2024-10-21): Switch back to default filename for exports after the active document changes. Fixes #16839
closed 2024-10-21 by yorikvanhavre