FreeCAD/FreeCAD #16841: TachDraw/BIM: Add the ability to specify elevation marks.

kaiwas, 2024-09-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This type of dimensions is often used in construction design. These dimensions allow you to draw a line along its entire length.
The values ​​are indicated in meters (with three decimal places). At the same time, the dimension itself is not indicated, because it is always in meters.

At the bottom of the picture you can see dimension lines corresponding to Russian standards. But other options are possible (I do not know all the current rules on the planet).
[image]

For example, in the Archicad program they look completely different and there are many different designs.
[image]

Another feature of these dimensions is the ability to place the arrow both down and up.

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38827 (Git) AppImage
Build type: Release
Branch: main
Hash: abd5c3ca31539f8ed37ac3dec2b9abd252a5310d
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * PitchedRoof
  * lattice2 1.0.0
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-26:
Hi @ebrahimraeyat do you have experience with Arch/BIM/TechDraw interactions ? Just wondered since you're projects OSAFE, SteelColumn, etabs might have some overlap ?

ebrahimraeyat, 2024-09-26:
Hi @luzpaz. Thanks for mention.
In my country, Iran, also we use this type of dimension that we called "LEADER", like this picture:

[image]

but I did not use those in my workbenchs, only when I export the drawing to DXF, I add it manually.

kaiwas, 2024-09-27:
> I add it manually

yes. The absence of this type of arrows is one of the main reasons for the impossibility of drawing up construction drawings in FreeCAD and it has to be done in other programs.

yorikvanhavre, 2024-09-30:
> yes. The absence of this type of arrows is one of the main reasons for the impossibility of drawing up construction drawings in FreeCAD and it has to be done in other programs.

It's totally possible to do these using Draft tools, though. I've been doing construction drawings with FreeCAD for a long time.

Having those kinds of Arch/BIM symbols directly in TechDraw would be part of a bigger move from the BIM->Draft->TechDraw paradigm to a pure BIM->TechDraw one. That's a pretty big and ambitious work, though, it depends heavily on TNP mitigation and it might not happen tomorrow.

kaiwas, 2024-09-30:
> Draft tools, though.

And what tools did you use to do this in the Draft workbench? Did you draw lines and use text?
There is no such type of dimensions as elevation marks there either.

yorikvanhavre, 2024-09-30:
I usually draw the marks myself with lines, polylines, etc. then make a compound of it, then clone that compound around.

Note that in TechDraw, you can also create your own SVG symbols, even with editable texts. So everything is there alreafy for you to make your own level mark tool...

kaiwas, 2024-09-30:
> I usually draw the marks myself with lines, polylines, etc. then make a compound of it, then clone that compound around.
> 
> Note that in TechDraw, you can also create your own SVG symbols, even with editable texts. So everything is there alreafy for you to make your own level mark tool...

This is a workaround, but it doesn't seem convenient. Hopefully things will change someday (not necessarily soon)

luzpaz, 2024-10-05:
> Note that in TechDraw, you can also create your own SVG symbols, even with editable texts. So everything is there alreafy for you to make your own level mark tool...

@yorikvanhavre maybe worth bundling FreeCAD with those symbols ? (So folks see a viable workaround till this FR is worked on ?)

yorikvanhavre, 2024-10-07:
> @yorikvanhavre maybe worth bundling FreeCAD with those symbols ? (So folks see a viable workaround till this FR is worked on ?)

Yes, having that kind of symbols as a convenient tool in FreeCAD would be excellent, the problem is that they are highly dependent on where you work, as you usually need to follow some national standard. I think the best would be to have a good library of those symbols that one could drop into FreeCAD. I tried to start one like that years ago ( https://github.com/FreeCAD/FreeCAD-symbols ) but it never took off...

kaiwas, 2024-10-07:
> some national standard.

In fact, regional standards do not differ that much. Yes, the sizes of elements may differ slightly, but the general principle is almost always the same.
In addition, you can ask the community for help. Let people show what elements are used in their countries. I think if the designer has experience, he will easily be able to provide an example and indicate the country where a certain solution is used.

kaiwas, 2024-10-07:
> but it never took off...

No wonder it didn't take off. I spent half an hour trying to run the macro to add objects, but I still couldn't do it (((

kaiwas, 2024-10-08:
I experimented with imported SVG objects in TechDraw. I think it is not worth wasting time on this at the moment. They are completely inoperative, because it is impossible to accurately position them using snap (there is no snap). The text also has to be written separately. At the moment, the TechDraw workbench can only be used to generate views and sections and export them for further work in other programs. The TechDraw workbench is completely useless for drawing up documentation (construction).