FreeCAD/FreeCAD #16842: Add identifier to tree view for objects marked as "Toggle selectability"

kaiwas, 2024-09-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

We can mark objects as "non-selectable", but there is no way to identify such objects in the view tree.

The only way to identify them is to try to select them. Then we will see which objects are not highlighted. In this case, it is best to select with a frame, because single clicks can highlight edges that are difficult to see. This is very inconvenient.

Video demonstration.

https://github.com/user-attachments/assets/860ec961-7be1-429b-bc0c-3f4934bc2464

I suggest adding some kind of label to the view tree for objects that are not available for selection.

[image]

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38827 (Git) AppImage
Build type: Release
Branch: main
Hash: abd5c3ca31539f8ed37ac3dec2b9abd252a5310d
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * PitchedRoof
  * lattice2 1.0.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

ToaKraka, 2024-10-05:
See also realthunder/FreeCAD_assembly3#945.

kaiwas, 2024-10-22:
> See also [realthunder/FreeCAD_assembly3#945](https://github.com/realthunder/FreeCAD_assembly3/issues/945).

Thanks, but what to do with it? I never used these builds and did not plan to do so.

luzpaz, 2024-10-22:
@kaiwas the link shows the code that Link branch used to implement it.

kaiwas, 2024-10-22:
> @kaiwas the link shows the code that is used to implement into Assembly3.

It sounds like a solution already exists, but it just needs to be moved to the main branch. ;)

luzpaz, 2024-10-22:
@kaiwas yes, that is what @ToaKraka was conveying from the start :wink: