FreeCAD/FreeCAD #16844: 1.0.0RC2 crash on shapebinder

lucapalazzo, 2024-09-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hi All,
RC2 crash every time I select a shapebinder (just select or try to open contextual menu).
I get in console:
`libc++abi: terminating due to uncaught exception of type Standard_TypeMismatch
Abort trap: 6
`
Luca

### Full version info

```shell
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
Installed mods: 
  * ThreadProfile 1.92.0
  * ThreadProfile.backup1719500048.0999238 1.89.0 (Disabled)
  * freecad.gears 1.0.0
  * BIM021 2021.12.0
  * FeedsAndSpeeds 0.5.0
  * woodworking 0.21.2.33771
  * Render 2022.4.0
  * WebTools 1.0.0
  * Pyramids-and-Polyhedrons
  * Manipulator 1.5.7
  * kicadStepUpMod 11.3.9
  * fasteners 0.4.52
  * sheetmetal 0.2.60
  * ExplodedAssembly
  * Curves 0.6.47
  * Assembly4 0.50.15
```

### Subproject(s) affected?

None

### Anything else?

[trace.txt](https://github.com/user-attachments/files/17147543/trace.txt)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

AIRCAP, 2024-09-27:
I think this depends on the type of shape that the shapebinder refers to. Can you post an example FCStd file or explain in more detail which shape binder you made of what kind of shape?

AIRCAP, 2024-09-27:
I can't reproduce the crash with arbitrary shapes - I tried a shape binder to a cylinder and one to a loft and also one with specific edges selected - neither of them caused a crash. in the backtrace Measure called Measure::Measurement::radius() - that suggests some kind of round object, but circular wires and cylinders didn't trigger this.
Please provide an example file.

hyarion, 2024-09-30:
https://github.com/FreeCAD/FreeCAD/pull/16906 will solve the macos specific crash, instead it will output an error message

lucapalazzo, 2024-10-01:
Here is example file that make FC100rc2 crash.
[FC100RC crash.FCStd.zip](https://github.com/user-attachments/files/17200317/FC100RC.crash.FCStd.zip)

hyarion, 2024-10-01:
> Here is example file that make FC100rc2 crash. [FC100RC crash.FCStd.zip](https://github.com/user-attachments/files/17200317/FC100RC.crash.FCStd.zip)

I cannot confirm the crash when loading the file in the UI.

FreeCAD is crashing when when double clicking a file in Finder on macos (comments available in the discussion in this issue https://github.com/FreeCAD/FreeCAD/issues/16828).

The only happens on application bundles (FreeCAD.app) and I'm waiting on a new weekly to check if the latest merged changes done to the splashscreen fixes this issue as suggested by bgbsww.

Roy-043, 2024-10-03:
> Here is example file that make FC100rc2 crash. [FC100RC crash.FCStd.zip](https://github.com/user-attachments/files/17200317/FC100RC.crash.FCStd.zip)

I get the same warning when I select a shapebinder in that file. But because a spreadsheet is missing the file won't recompute properly. I do not know if that is relevant.

azuk, 2024-10-03:
> I get the same warning when I select a shapebinder in that file. 

"Me too". The "unknown" exception that is thrown is actually Standard_TypeMismatch. It is thrown by OCCT code. Here's the stacktrace of the throwing situation:

```
#0  0x000015554dabb0f1 in __cxa_throw () from /lib/x86_64-linux-gnu/libstdc++.so.6
#1  0x00001555219b6169 in TopoDS::Face (S=...) at /usr/local/occt/include/opencascade/TopoDS.lxx:119
#2  0x00001555219b2105 in Measure::Measurement::radius (this=0x555556fe3c40)
    at /home/foo//freecad/src/Mod/Measure/App/Measurement.cpp:584
#3  0x000015551498b4e3 in MeasureGui::QuickMeasure::printResult (this=0x55555701e9b0)
    at /home/foo//freecad/src/Mod/Measure/Gui/QuickMeasure.cpp:190
#4  0x000015551498abde in MeasureGui::QuickMeasure::tryMeasureSelection (this=0x55555701e9b0)
    at /home/foo//freecad/src/Mod/Measure/Gui/QuickMeasure.cpp:101
#5  0x000015551498aafd in MeasureGui::QuickMeasure::processSelection (this=0x55555701e9b0)
    at /home/foo//freecad/src/Mod/Measure/Gui/QuickMeasure.cpp:81
#6  0x000015551498e41d in QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (MeasureGui::QuickMeasure::*)()>::call(void (MeasureGui::QuickMeasure::*)(), MeasureGui::QuickMeasure*, void**) (
    f=(void (MeasureGui::QuickMeasure::*)(class MeasureGui::QuickMeasure * const)) 0x15551498aac2 <MeasureGui::QuickMeasure::processSelection()>, o=0x55555701e9b0, arg=0x7fffffffc8b0)
    at /usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h:152
#7  0x000015551498e290 in QtPrivate::FunctionPointer<void (MeasureGui::QuickMeasure::*)()>::call<QtPrivate::List<>, void>(void (MeasureGui::QuickMeasure::*)(), MeasureGui::QuickMeasure*, void**) (
    f=(void (MeasureGui::QuickMeasure::*)(class MeasureGui::QuickMeasure * const)) 0x15551498aac2 <MeasureGui::QuickMeasure::processSelection()>, o=0x55555701e9b0, arg=0x7fffffffc8b0)
    at /usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h:185
#8  0x000015551498e031 in QtPrivate::QSlotObject<void (MeasureGui::QuickMeasure::*)(), QtPrivate::List<>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) (which=1, this_=0x55555701f5c0, r=0x55555701e9b0, 
    a=0x7fffffffc8b0, ret=0x0) at /usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h:418
#9  0x000015554e112e16 in ?? () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#10 0x000015554e11710d in QTimer::timeout(QTimer::QPrivateSignal) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#11 0x000015554e10624b in QObject::event(QEvent*) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#12 0x000015554ed6bd45 in QApplicationPrivate::notify_helper(QObject*, QEvent*) ()
   from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#13 0x0000155552cc558a in Gui::GUIApplication::notify (this=0x7fffffffd410, receiver=0x555557017cd0, 
    event=0x7fffffffcc80) at /home/foo//freecad/src/Gui/GuiApplication.cpp:93
#14 0x000015554e0d8118 in QCoreApplication::notifyInternal2(QObject*, QEvent*) ()
   from /lib/x86_64-linux-gnu/libQt5Core.so.5
#15 0x000015554e1345ab in QTimerInfoList::activateTimers() () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#16 0x000015554e134ed9 in ?? () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#17 0x000015554c3145b5 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#18 0x000015554c373717 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#19 0x000015554c313a53 in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#20 0x000015554e135279 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) ()
   from /lib/x86_64-linux-gnu/libQt5Core.so.5
#21 0x000015554e0d6a7b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) ()
   from /lib/x86_64-linux-gnu/libQt5Core.so.5
#22 0x000015554e0df3e8 in QCoreApplication::exec() () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#23 0x0000155552b16c37 in (anonymous namespace)::tryRunEventLoop (mainApp=...)
    at /home/foo//freecad/src/Gui/Application.cpp:2207
#24 0x0000155552b16f90 in (anonymous namespace)::runEventLoop (mainApp=...)
    at /home/foo//freecad/src/Gui/Application.cpp:2227
#25 0x0000155552b1736e in Gui::Application::runApplication ()
    at /home/foo//freecad/src/Gui/Application.cpp:2304
#26 0x000055555557acc2 in main (argc=1, argv=0x7fffffffdb58)
    at /home/foo//freecad/src/Main/MainGui.cpp:295
```

lucapalazzo, 2024-10-08:
I can confirm that latest weekly build does not crash anymore.
Thanks

hyarion, 2024-10-20:
@maxwxyz it looks like this has been resolved and op is happy, should we close it?

maxwxyz, 2024-10-20:
thanks for the ping

closed 2024-10-20 by maxwxyz