FreeCAD/FreeCAD #16846: VarSet properties should not permit reserved names

kriswilk, 2024-09-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It is currently possible to use reserved names for VarSet properties (perhaps elsewhere in FC, too?). For instance, unit names like `T`, `K`, and `rad` are permitted without any error/warning. Furthermore, the expression auto-completer also correctly offers them. However, the expression parser will report `failed to parse expression`.

**DEMO** - create a VarSet with length property `T`, then try to dimension a sketch line with `<<VarSet>>.T`

Even for a proficient user (but especially for a new one) this behavior is cryptic. The problem is compounded by the fact that properties cannot be renamed... the user must delete and recreate the property with a new, hopefully-valid name.

FreeCAD should not permit such name collisions.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

None

### Anything else?

**The impact of this bug should not be underestimated.**

I came across it while showing a beginner how VarSets work. To make a parametric circular plate I created sensible properties `D` (diameter) and `T` (thickness). Property `D` worked fine, but when trying to pad my sketch with length `T`, it failed. I really had no idea why until I discovered that `T` is a reserved unit name.

Many users are likely to choose single letters for property names as they first experiment with this feature. Hitting a mysterious brick wall is not a good introduction.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-09-27:
@pieterhijma FYI

kriswilk, 2024-09-27:
First, thanks @pieterhijma for this handy feature. I wanted to add two points:

- Because of the extra scrutiny that new features in 1.0 will get, I suggest that this bug be prioritized for a fix in RC3. Not sure if it qualifies as a blocker but it sure is easy to run into.
- To speed up a fix, refer to aliases in the Spreadsheet workbench (reserved names are handled there). I suspect some of that code can be reused or refactored for this situation.

mwganson, 2024-09-28:
In seeing this I realized DynamicData suffers the same issue.  I have fixed it in version 2.64.  My method in python was to use this function:

```
    def isUnit(self, name):
        """check if name is a reserved keyword for units, such as T or k"""
        #if parsing quantity succeeds, it means this name is a reserved keyword
        try:
            FreeCAD.Units.parseQuantity(name)
            return True
        except:
            return False
```

In c++, look at how the spreadsheet validation is done:

```
    /* Check to make sure it doesn't clash with a predefined unit */
    if (ExpressionParser::isTokenAUnit(candidate)) {
        return false;
    }

```

https://github.com/FreeCAD/FreeCAD/blob/6e2cd4e73303f817e4bf5e908abe39030b7c5fb3/src/Mod/Spreadsheet/App/PropertySheet.cpp#L130

kriswilk, 2024-09-28:
> In c++, look at how the spreadsheet validation is done:
> 
> ```
>     /* Check to make sure it doesn't clash with a predefined unit */
>     if (ExpressionParser::isTokenAUnit(candidate)) {
>         return false;
>     }
> ```

Glad you were able to fix this in DynamicData, @mwganson. 👍

I also found the reference to `ExpressionParser::isTokenAUnit()`. I've prepared what I think is the necessary fix for both for VarSets and in the naming of custom properties in general (which I discovered also overlook reserved names!). Need to compile and test it this evening before submitting a PR.

From what I can see, there are three locations where this check needs to be added:

1) The general "Add Property" dialog box:
https://github.com/FreeCAD/FreeCAD/blob/6e2cd4e73303f817e4bf5e908abe39030b7c5fb3/src/Gui/DlgAddProperty.cpp#L96

2) VarSet's specific "Add Property" dialog box:
https://github.com/FreeCAD/FreeCAD/blob/6e2cd4e73303f817e4bf5e908abe39030b7c5fb3/src/Gui/DlgAddPropertyVarSet.cpp#L343

3) The expression input dialog box (where VarSet property creation can also occur):
https://github.com/FreeCAD/FreeCAD/blob/6e2cd4e73303f817e4bf5e908abe39030b7c5fb3/src/Gui/DlgExpressionInput.cpp#L411

Once submitted, I hope @pieterhijma and/or others more experienced than me can double-check it and get it merged in time for 1.0.

commit ffa11896 (2024-10-03): Prevent using reserved names for properties or aliases (Fix #16846) (#16902)

* Prevent naming properties with reserved words (Fix #16846)

* Prevent using reserved constant names for properties or aliases

* [pre-commit.ci] auto fixes from pre-commit.com hooks

for more information, see https://pre-commit.ci

* App: Add unit test for isTokenAConstant

---------
closed 2024-10-03 by chennes