FreeCAD/FreeCAD #16849: Generation of unique names is a performance sinkhole due to poor algorithms

kpemartin, 2024-09-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I recently observed code in `Document.addObject` which appeared to consume time proportional the the square of the number of objects in the drawing.

To investigate this further I made a sample DXF file containing 20,000 LINE objects and imported it. Note that although this problem was demonstrated using the DXF importer, it is not per se a problem with this importer and so should not be tagged as such.

The import operation took about 5 minutes, of which about 70% was spent generating unique names and labels for the objects.

Soon after the import, an auto-save started, and this froze the app for about 40 minutes. 99% of this time was spent generating unique names for "files" within the compound save-file format. It turns out it needs 5 of these for each of the original LINE objects, so it is generating 200,000 unique names in a single namespace.

Granted, this example is a fairly large document but not unrealistically so, and its size serves to emphasize such inefficiencies, which lead not only to slowness but often also to heap fragmentation and memory bloat.

The problem originates in `Tools.getUniqueName` which takes a `vector<string>&` as a collection of names that already exist. This means that `getUniqueName` must do linear searching to detect name clashes, and also that the caller must prepare this vector to pass to the method. Both these operations are O(n) on the number of names already present, so overall creating a name collection holding n names takes O(n^2) time.

### Full version info

```shell
[code]
OS: Windows 7 Version 6.1 (Build 7601: SP 1)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Anything else?

I'll look into creating a suitable class for name management with unique-name generation. Ideally it will be able to replace existing child-object containers rather than having to exist side-by-side merely as a name manager.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

sslupsky, 2024-09-26:
You appear to be running a very old version of FreeCAD (0.20) ... have you looked at a more recent version 1.0.0RC1?  There was a great deal of work done with respect to TNP so it is possible this has been affected.

kpemartin, 2024-09-26:
This occurs with every version I've tried including one freshly-built from the master branch of the source.

What is "TNP" ?

sslupsky, 2024-09-26:
TNP = Topological Naming Problem,  There was a lot of change to FreeCAD this year to integrate the changes to address TNP so I was wondering if your issue was captured by the changes.

luzpaz, 2024-09-26:
https://wiki.freecad.org/Topological_naming_problem

mnesarco, 2024-10-03:
This is an interesting issue. I can confirm this is not related to TNP at all. And my first light review of the code shows that @kpemartin has a point, the existing code is highly inefficient. 

@kpemartin I like the idea of a dedicated class for name management. It appears to some "unique" names are scoped, for example unique property names are unique in its parent scope. Maybe the name manager should care only about File, Document and Object names. Object Names are unique within its document, so there is scope anyway.

kpemartin, 2024-10-04:
I'm still working on it. I have a dedicated class written for name management, and I'm testing it for the Document's collection of DocumentObjects. Unfortunately the class is a template class and so is coded within an include file (currently in Document.h but once it is working I'll move it somewhere more appropriate) which means every time I change it a full rebuild is necessary.

Makes for slow progress.

The next place to use it would be the Writer's "file" collection as this is the big time suck in auto-save (and likely in real document save as well) when there are many objects in the document.

I don't think it would be necessary for property names. These tend to be a small and finite set, and although the names must be unique, this (making names unique) does not appear to be done at runtime, or if it is, it is done through a mechanism other than `Tools.getUniqueName` so if you can point me to a specific instance of this I can look into it later.

mnesarco, 2024-10-04:
@kpemartin i agree on leaving property names apart. Apparently it is used here (for reference): https://github.com/FreeCAD/FreeCAD/blob/32e09d9554db26414697ed19e0189ccab4ea9029/src/App/DynamicProperty.cpp#L277

kpemartin, 2024-10-04:
Ah, yes, that's what I get for working from memory...

I will probably change this even though efficiency isn't much of a concern here because I want to eliminate `Base::Tools::getUniqueName`. Its mere existence is problematic since its usage model encourages poor efficiency.

closed 2025-02-24 by chennes