FreeCAD/FreeCAD #16851: Techdraw: Adding measurements does not work anymore between lines and lircle center line.

ostetzer, 2024-09-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This used to work in previous versions and does not work with current RC2:
When I try to add a horinzontal or vertical measurement between an edge and a (parallel) hole center line the following error message prevents this:

Can not make angle from parallel edges

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Switzerland (de_CH)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * A2plus 0.4.60n
  * ArchTextures
  * Assembly4 0.50.12
  * CADExchanger
  * Defeaturing 1.2.2
  * DynamicData 2.62.0
  * fasteners 0.5.21
  * Help 1.0.3
  * LCInterlocking
  * parts_library
  * woodworking 0.21.2.33771
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-09-26:
I can't reproduce this. Can you attach a file and specify which tool you use?

ostetzer, 2024-09-26:
Hmmm FCStd file type is not supported here for uploading.
Its a file I worked at since about a year with previous versions and imported it into the new 1.0 version.
If that helps: I have used A2plus assembly and had several LCS in the file that I recently deleted.

I may try adding the file to zip for uploading...

ostetzer, 2024-09-26:
[Z_Platte.zip](https://github.com/user-attachments/files/17155380/Z_Platte.zip)

I also created a new small demo file 1 extrusion, 1 hole and here the error did not occur.
Interesting: In the new file the holes had already hole center lines by default, in my file I had to add them manually.
What does work in my file: Adding a measurement between two hole center lines.

Roy-043, 2024-09-27:
I can confirm the dimension issue. But if I create a new view the problem does not occur.
I can also confirm the centerline issues (also for the view in the attached file).

[Z_Platte-roy.FCStd.zip](https://github.com/user-attachments/files/17161369/Z_Platte-roy.FCStd.zip) (remove .zip)

```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38830 (Git)
Build type: Release
Branch: main
Hash: b38c42262f58cf0587b627e7a866d3756da34dc4
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```

WandererFan, 2024-09-29:
I can't duplicate this.  Are we using the individual tools or the combined "super dimension tool"?

ostetzer, 2024-09-29:
One observation that might help:
In the "old" view that I used where the error occured, the holce center crossline consisted of two individual perpendicular lines which were individually selectable.
When I create a new view (where the error does not occur) the hole center lines are automatically generated and the cross is only one clickable object.

ostetzer, 2024-10-02:
Hi, I don't know if this related or a new bug:
When I export the page as pdf, the hole center lines are not exported and missing in the pdf.
(also, but definitely another bug: Printing directly does not work, I get the message "template not set for page").

commit 528c2a0a (2024-10-07): [TD]fix parallel edge test (fix  #16851)

- missing normalization before test
commit 32c664bd (2024-10-07): [TD]fix parallel edge test (fix  #16851)

- missing normalization before test
closed 2024-10-07 by yorikvanhavre