FreeCAD/FreeCAD #16854: Importing common airfoil data does not work (Freecad 1.0RC2)

stefanpeter218, 2024-09-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Trying to import common airfoil data does not work. The resulting error message is:
22:33:22  pyException: Traceback (most recent call last):
  File "/tmp/.mount_FreeCAUAC4Ig/usr/lib/python3.11/site-packages/freecad/module_io.py", line 12, in OpenInsertObject
    except PyExc_FC_AbortIOException:
           ^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'NameError'>: name 'PyExc_FC_AbortIOException' is not defined

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Estimate 0.1.2
  * AirPlaneDesign 0.4.0
  * Curves 0.6.40
  * Manipulator 1.5.7
  * CurvedShapes 1.0.9
  * OpenDark 2024.3.13
```

### Subproject(s) affected?

None

### Anything else?

[sample-airfoil-data.zip](https://github.com/user-attachments/files/17155794/sample-airfoil-data.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-27:
Being addressed in #16847  
Closing as duplicate. Please try to search for the error next time. Thanks for reporting nonetheless.

luzpaz, 2024-09-27:
Re-opening. Should be tested before closing.

hyarion, 2024-09-29:
[image]

It does load something after #16847, but I have some swig issues, so I can't verify that everything works as it should.

Syres916, 2024-10-09:
Opens successfully (error free) for me using:

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38926 (Git)
Build type: Release
Branch: main
Hash: a9bff78974f301100134a13c636e23aa91be5bcf
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Installed mods: 
  * FreeCAD-themes 2024.9.14
  * Silk 0.1.5
  * sheetmetal 0.4.24
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.27.5
  * 3dfindit-freecad-integration 1.2.0
  * fasteners 0.5.29
  * freecad.gears 1.3.0
  * Curves 0.6.48
  * A2plus 0.4.68
```

[image]

closed 2024-10-16 by luzpaz