FreeCAD/FreeCAD #16856: PartDesign: Loft: closed ruled lofts between 2 wires become buckets in 1.1.0dev vs shells in 0.21

AIRCAP, 2024-09-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

(Potential Regression)

In 1.1.0dev (with new TopoShape code) the following file
[weird.zip](https://github.com/user-attachments/files/17156400/weird.zip)
becomes a bucket (partdesign loft between 2 sketches with property "ruled surfaces" on and "closed" set to true
[image]

in 0.21 (before refactoring) its a hollow shape
[image]

### Full version info

```shell
OCC 7.6.3 for both - release 0.21.2 vs latest main ( 69676a0f9d2d5b288bda3d1c6dbad5a6ec26bfaa )
```

### Subproject(s) affected?

PartDesign

### Anything else?

skipped irrelevant information from the bug report, it is known which commits introduced the new loft implementation and where to find that code (FeatureExtrude.cpp)

The question is , does this need fixing? Who ever actually uses ruled or closed lofts - considering the feature did and does pretty much the opposite?

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

AIRCAP, 2024-09-26:
from what i hear both old and new behaviour is "wrong" as part design should always create solids - and the "closed" flag in the loft is a code artefact in the beginning that should not be there...

FlachyJoe, 2024-10-07:
Closed is ok, it uses the same profile for start and end and create a loop, but it only works for 3 and more profiles.

PR is coming.

AIRCAP, 2024-10-07:
cool

PR #17080 (2024-10-14): [PartDesign] Disable closed Loft for less than 3 profiles
Fix #16856
closed 2024-10-14 by yorikvanhavre