FreeCAD/FreeCAD #16861: Expression / Formula editor doesn't respect units.

Connor9220, 2024-09-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If you go into a parameter box and click the formula / expression icon in the upper right hand corner and type in a number without a unit (such as in or " for example), it returns back with the proper number in the main parameter box, however, when you okay out, it converts the number to metric.  If you simply type in the number in the normal parameter box without going into the expression / formula editor, it respects the default units as expected.  This looks to affect multiple workbenches including sketcher and CAM.

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * OpenDark 2024.3.13
  * lattice2 1.0.0
  * ThreadProfile 1.92.0
  * fasteners 0.5.28
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

benj5378, 2024-10-04:
For me, putting a Spreadsheet cell `=(5 mm)^2)` is given as `6,25 cm^2` on my non-wayland, it's given as `25 mm^2`.

We might also mention #13534 and #12173... idk if #11862 is Expression or Spreadsheet issue

tritao, 2026-08-29:
I looked into this briefly as part of the hackaton, since I'm already looking into a sourrounding area in https://github.com/FreeCAD/FreeCAD/pull/30139.

This is an analysis that explains the issue and how we can fix it cleanly as part of a follow-up PR:
https://gist.github.com/tritao/6610fe63917cd1651d2e1e95c5254849

pieterhijma, 2026-08-29:
Great, thanks, @tritao!