FreeCAD/FreeCAD #16866: Sketcher: carbon copy from another sketch can't be modified

Panda631, 2024-09-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When a sketch is carbon copied from another sketch, the new can't be modified. It follows the first sketch like a clone.
Can't imagine that this is the designed behavior?

See attached project.
Sketch001 is a carbon copy from sketch

[image]

Thuis is how sketch001 looks like

[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * EM 2.1.1
  * fasteners 0.5.27
  * Help 1.0.3
  * kicadStepUpMod 10.17.1
  * Pyramids-and-Polyhedrons
  * Render 2024.8.4
  * sheetmetal 0.2.61
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-09-27:
The user has the option to change the geometry to construction after creation and has been a successful workflow for me for at least 8 years. When using the Master sketch workflow the user makes the base sketch as simple as possible and uses subsequent carbon copies to be more complex building _upon_ the base geometry. The subsequent carbon copies can also be rotated and re-positioned to where the user requires if for example the base geometry was a cross section.

kaiwas, 2024-09-27:
I can't confirm. The copies are completely independent and are connected only by a constraint.

In the video, geometry from the Sketch (black) was copied into Sketch001 (red)

https://github.com/user-attachments/assets/baba0a6b-6290-43ec-8d93-98e03975dd28

By the way, I personally would prefer that if new geometry is added to the copied sketch, this geometry would be added to another sketch. But it seems that carbon copy is a one-time operation.

Panda631, 2024-09-27:
Here example project shown in my original post.

[blueprint.zip](https://github.com/user-attachments/files/17164203/blueprint.zip)

Panda631, 2024-09-27:
> I can't confirm. The copies are completely independent and are connected only by a constraint.
> 
> In the video, geometry from the Sketch (black) was copied into Sketch001 (red)
> vokoscreenNG-2024-09-27_15-27-43.mp4
> 
> By the way, I personally would prefer that if new geometry is added to the copied sketch, this geometry would be added to another sketch. But it seems that carbon copy is a one-time operation.

Did you use copy & past?
This works as expected. carbon copy did not.

kaiwas, 2024-09-27:
> Did you use copy & past?
> This works as expected. carbon copy did not.

No. I used carbon copy. But there is one key feature - I did not use Part Design )

Your file has a symmetry contour relative to the origin. Maybe this is the issue?

Panda631, 2024-09-27:
> > Did you use copy & past?
> > This works as expected. carbon copy did not.
> 
> No. I used carbon copy. But there is one key feature - I did not use Part Design )
> 
> Your file has a symmetry contour relative to the origin. Maybe this is the issue?
> vokoscreenNG-2024-09-27_15-43-34.mp4
> 
> I was interested in the question of how you made duplicate constraints? ))) ![Scr_24-09-27-03-45-26](https://private-user-images.githubusercontent.com/32273833/371536984-c76b880a-9fbc-453e-ad37-7b909dffd8f2.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjc0NDE4OTksIm5iZiI6MTcyNzQ0MTU5OSwicGF0aCI6Ii8zMjI3MzgzMy8zNzE1MzY5ODQtYzc2Yjg4MGEtOWZiYy00NTNlLWFkMzctN2I5MDlkZmZkOGYyLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTI3VDEyNTMxOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTYzMGFlMmZjY2EyNGJjNDI2NDcxOTYzNTc3MzQxN2NmNjlhYWZjNzYyOTdhYTRiYmRlNjI4MzJkMmQzZTIyYTQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.50tpqH4zt3X0b6aIHrwXe61pi-uI9uUppEx_AwExBIQ)

I didn't do this intentionally. This a effect of carbon copy.

maxwxyz, 2024-09-28:
So is this issue actually working as expected or a bug?