FreeCAD/FreeCAD #16868: Midplane + "Two lengths" = broken state

masterxq, 2024-09-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If you select midplane in a pad, and then switch over to "two lengths", the result will be a symmetric pad with 1 length.
I did see this problem the first time in 1.0.0. And it definitly changed the behavior from 0.21.
This is specially a problem when migrating documents from an older Version, but will also be annoying for beginners as "two lengths" not acts as expected and it is hard to guess that you need to go back to another type, disable midplane and then select "two lengths" again.

[image]

### Full version info

```shell
OS: Debian GNU/Linux trixie/sid (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * fcgear 1.0.0
  * kicadStepUpMod 10.18.6
  * Curves 0.6.8
  * Dracula 0.0.5
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-09-28:
Isn't this the same as https://github.com/FreeCAD/FreeCAD/issues/16072 ?
@chennes @Syres916 should we label this a blocker? PR https://github.com/FreeCAD/FreeCAD/pull/16087 was closed, not merged

masterxq, 2024-10-07:
> Isn't this the same as #16072 ?

Yes I think this is the same issue.
Fell free to close one of them :)

maxwxyz, 2024-10-08:
Closing as duplicate

closed 2024-10-08 by maxwxyz