FreeCAD/FreeCAD #16876: Feature Request: A right angle tool in sketcher.

adl2494, 2024-09-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I apologize if this has already been suggested, however I did search and was not able to locate the idea anywhere.  I think it could be helpful to add to the Sketch_ConstrainAngle Button, a menu option to include a Sketch_ConstrainRightAngle.  Obviously this would not have to prompt for the angle, and It would be nice if the resulting visual constraint was a simple corner square as opposed to the angle showing degrees on the drawing, which can sometimes be cluttering.  This is obviously a small thing, however in my mind, seems simple to implement.  However, my coding ability is quite limited.  Thank you.  

### Full version info

```shell
[code]
OS: Windows 7 build 7601
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * 3DfindIT 1.2.0
  * Assembly4 0.50.15
  * DynamicData 2.62.0
  * fasteners 0.4.66
  * Help 1.0.3
  * kicadStepUpMod 10.23.0
  * pcb 6.2023.1
[/code]
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-09-27:
[Sketcher ConstrainPerpendicular](https://wiki.freecad.org/Sketcher_ConstrainPerpendicular) ?

adl2494, 2024-09-27:
Exsists

closed 2024-09-27 by adl2494