FreeCAD/FreeCAD #16884: Measurement Label will stay and have no handle to delete it

masterxq, 2024-09-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I'm very unsure if this could be a duplication, at least this should be checked #16829. But I think this is something else.

One way to reproduce one of the problems is:
1. Open the measurement tool
2. make a measurement
3. Press reset
4. Close the view
5. Messurement will stay and not appear in the tree view

[image]

[image]

Restart FreeCAD will remove the label.

There are really many bugs in the measurement tool, it should be tested by a developer and all problems should be solved ASAP, you will create a lot of management overheads with all the bug tickets that will be created if this is not cleaned up ASAP.

### Full version info

```shell
OS: Debian GNU/Linux trixie/sid (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * fcgear 1.0.0
  * kicadStepUpMod 10.18.6
  * Curves 0.6.8
  * Dracula 0.0.5
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-09-28:
Cannot confirm with 
```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.27.10
  * Curves 0.6.47
  * dodo 1.0.1
  * fasteners 0.5.29
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.24
  * Rocket 3.3.0
  * sheetmetal 0.4.26
  * Silk 0.1.5
  * Silk.backup1727166062.7227929 0.1.5 (Disabled)
```

FEA-eng, 2024-09-28:
Yeah, I also have it and it seems to be addressed by #16836.

closed 2024-09-30 by yorikvanhavre