FreeCAD/FreeCAD #16885: Part/Sketcher: dependent object's draw style messed up after editing sketch

jffmichi, 2024-09-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After editing a sketch some dependent objects' draw style get messed up. The objects are rendered without any lighting (see attached picture). Restarting FreeCAD solves the issue. Changing the draw style manually only sometimes solves the issue. Changing the object's appearance also only sometimes solves the issue.

Steps to reproduce:

1) open attached file [sketch_light_issue.zip](https://github.com/user-attachments/files/17172809/sketch_light_issue.zip)
2) double-click "Sketch"
3) close sketch editor

### Full version info

```shell
OS: Arch Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38742 (Git)
Build type: Release
Branch: main
Hash: b6599c5315ba557371dbdacca0b9cf35f5b7f7f2
Python 3.12.6, Qt 6.7.2, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.2.dev
Locale: English/United States (en_US) [ OS: German/Germany (de_DE) ]
Stylesheet/Theme/QtStyle: unset/Classic/breeze
Installed mods: 
  * kicadStepUpMod 11.2.7
  * dodo 1.0.1
  * sheetmetal 0.4.24
  * fcgear 1.0.0
  * fasteners 0.5.27
  * Assembly4 0.50.15
  * Curves 0.6.45
  * lattice2 1.0.0
```

### Subproject(s) affected?

Sketcher

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-09-28:
@CalligaroV you've made a recent PR, any idea?

depthoffocus, 2024-09-30:
I think this is the issue I raised as #15352.

I outlined some settings and steps that will trigger it here: 

https://github.com/FreeCAD/FreeCAD/issues/15352#issuecomment-2263743923

CalligaroV, 2024-10-06:
Hi @maxwxyz!

If you're referring to #16378 then I would say that at least is related but I'm not sure if it's the root cause.
Rather, after doing several local tests and digging deeper in the `ViewProviderSketch::setEdit()` / `ViewProviderSketch::setEditViewer()` / `ViewProviderSketch::unsetEdit()` /  `ViewProviderSketch::unsetEditViewer()` / `Gui::View3DInventorViewer::updateOverrideMode()` / `Gui::View3DInventorViewer::setOverrideMode()` I would say that my previous PR exposed a (probably small) issue.

Regarding what @jffmichi found I can say first that I can confirm this issue, I can reproduce it and I also probably found a way to fix it (moving the logic related to `disableShadedView` from `ViewProviderSketch::unsetEdit()` to `ViewProviderSketch::unsetEditViewer()`. This will make #16378's check redundant as the argument of `ViewProviderSketch::unsetEditViewer()` is already an Inventor / 3D View).
Probably I'll open a PR later today after finishing some local tests, or simply I'll push the modification in a branch on my fork here on GH and paste the link in this conversation.

I also agree with @depthoffocus: the combination of `User parameter:BaseApp/Preferences/Mod/Sketcher/General/DisableShadedView` and `User parameter:BaseApp/Preferences/Mod/Sketcher/General/HideDependent` (that sets the value of `ActiveSketch.ViewObject.HideDependent` and therefore executes / doesn't execute https://github.com/FreeCAD/FreeCAD/blob/32e09d9554db26414697ed19e0189ccab4ea9029/src/Mod/Sketcher/Gui/ViewProviderSketch.cpp#L3085) has effect on how the model is displayed after exiting Sketcher edit mode.
However, AFAICS after my tests, the only combination of preferences that actually triggers this issue is the same @depthoffocus described in the OP of #15352, that is:
 - `User parameter:BaseApp/Preferences/Mod/Sketcher/General/DisableShadedView` **enabled**
 - `User parameter:BaseApp/Preferences/Mod/Sketcher/General/HideDependent` **disabled**

I tried to figure out how to solve also that issue (which BTW can be reproduced also with the file attached in the OP of this one) by moving the logic related to `disableShadedView` in `ViewProviderSketch::[un]setEditViewer()` or `ViewProviderSketch::[un]setEdit()` but none of those modifications helped.

I also tested by commenting out the calls to `viewer->updateOverrideMode()` and `viewer->setOverrideMode()` in `src/Mod/Sketcher/Gui/ViewProviderSketch.cpp` (calling only the `update` or the `set` methods) and tried to modify the logic in `View3DInventorViewer::updateOverrideMode()` and `View3DInventorViewer::setOverrideMode()` but the outcomes are always the same, no fix for that issue with that specific combination of Parameters/Preferences.

---
**Conclusion:**
ATM I may have a fix for this specific issue but not for #15352

---
**(Maybe obvious) Side note:**
The value of `ActiveSketch.ViewObject.HideDependent` is the one that is saved in the .FCStd file, not the value of `User parameter:BaseApp/Preferences/Mod/Sketcher/General/HideDependent`.

To perform tests with different values of `ActiveSketch.ViewObject.HideDependent` **on already existing sketches** it is mandatory first to push the button `Apply to existing sketches` in the Preference Page `Sketcher->Display`

I did some test modifications and found that adding something like `pObserver->initParameters()` in `ViewProviderSketch::setEdit()` allows to do the tests above without going in the Preference Pages, but AFAICS this will actually overwrite the value of `ActiveSketch.ViewObject.HideDependent` that will be saved in the .FCStd file (together with other Properties, which may be a thing that users don't want) and anyway I guess that if this haven't been already added, while the button in the Preference Page have been added, is because this topic have been discussed and agreed that is the best thing to do.

closed 2025-02-24 by chennes