FreeCAD/FreeCAD #16886: STEP import fails

vovodroid, 2024-09-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Import of certain STEP files fails with error.

Import settings
[image]

File with Cyrillic name
[Ножки.zip](https://github.com/user-attachments/files/17173891/default.zip)

Another STEP:
[playground1.zip](https://github.com/user-attachments/files/17173902/playground1.zip)

Both fail with error
```
12:28:45  pyException: Traceback (most recent call last):
  File "C:\DISK_V\3D\FreeCAD_1.0.0RC2-conda-Windows-x86_64-py311\bin\Lib\site-packages\freecad\module_io.py", line 12, in OpenInsertObject
    except PyExc_FC_AbortIOException:
           ^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'NameError'>: name 'PyExc_FC_AbortIOException' is not defined
```

### Full version info

```shell
OS: Windows 10 build 19044
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * CurvedShapes 1.0.10
  * Curves 0.6.13
  * ThreadProfile 1.87.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-09-28:
does this work in the stable 0.21.2 version?

maxwxyz, 2024-09-28:
I can confirm, it won't let me import one or multiple STEP files.
[image]

```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.27.10
  * Curves 0.6.47
  * dodo 1.0.1
  * fasteners 0.5.29
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.24
  * Rocket 3.3.0
  * sheetmetal 0.4.26
  * Silk 0.1.5
  * Silk.backup1727166062.7227929 0.1.5 (Disabled)

```

maxwxyz, 2024-09-28:
@supermixed you've made a change last week https://github.com/FreeCAD/FreeCAD/pull/16499 could this be the reason?

vovodroid, 2024-09-28:
> does this work in the stable 0.21.2 version?

Yes, both files are open.

Syres916, 2024-09-28:
@maxwxyz this should be closed and linked to https://github.com/FreeCAD/FreeCAD/issues/16743 it's all the same root cause, non-ascii characters on any import fail, hopefully the various PRs will get it all back to normal on Monday evening.

vovodroid, 2024-09-28:
> on-ascii characters on any import fail,

Second step file name is ascii.

bgbsww, 2024-09-28:
Yeah, there are two issues in that commit - one is the non-ascii character one, and the other is the exception name.  Second file is triggering the exception, and the bad name is obscuring whatever the error actually is.  We don't yet know if it is legitimate.

maxwxyz, 2024-09-28:
I'll leave it open then. Which PR should close #16743 ?

narical, 2024-09-28:
Can  confirm, came here to leave a bugreport about exact this issue. Renaming step file helps.

maxwxyz, 2024-09-29:
So renaming my step file does not work, same issue.
[Stange.zip](https://github.com/user-attachments/files/17179308/Stange.zip)

bgbsww, 2024-09-29:
Huh.  No issue here.  You're on a build with the patch in it?

[image]

OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38847 (Git)
Build type: Debug
Branch: bgbsww-Multithreading
Hash: 9803a852af111e2fb099a8d4a5e02a6bfba0778f
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * sheetmetal 0.4.26
  * Curves 0.6.47
  * DynamicData 2.63.0
  * BIM021 2021.12.0
  * parts_library
  * Render 2024.9.25

maxwxyz, 2024-09-29:
no, without the patch. But I thought renaming the file should work too.

bgbsww, 2024-09-29:
I would have thought so, but there are lots of changes here.  Regardless, I think I'd go with Syres916's suggestion: https://github.com/FreeCAD/FreeCAD/issues/16886#issuecomment-2380624892

hyarion, 2024-09-29:
> Yeah, there are two issues in that commit - one is the non-ascii character one, and the other is the exception name. 

The exception name is handled in this PR: https://github.com/FreeCAD/FreeCAD/pull/16847

hyarion, 2024-10-03:
I've tried the files on latest main branch and the attached files work.

```
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38888 (Git)
Build type: Release
Branch: main
Hash: d423795e2dc32dc16f1e8c1835104deb01c9cab9
Python 3.11.8, Qt 5.15.8, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * A2plus 0.4.68
  * lattice2 1.0.0
  * CurvedShapes 1.0.10
  * sheetmetal 0.4.26
  * Curves 0.6.6
```

@maxwxyz I think this can be closed

closed 2024-10-03 by maxwxyz