FreeCAD/FreeCAD #16887: Unable to select a sketch

kachurovskiy, 2024-09-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FreeCAD RC1 doesn't allow to select any sketch in the Model view, rings an error. This is a regression compared to 21.2, there the file can be edited and fixed.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.60n
  * dxf-library
  * parts_library (Disabled)
```

### Subproject(s) affected?

None

### Anything else?

[BF10Mini.zip](https://github.com/user-attachments/files/17174419/BF10Mini.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-09-28:
I cannot reproduce, everything looks normal. Can you try without the config files (user.cfg).
What is the error when you try to select what? Can you provide a screenshot or video?

kachurovskiy, 2024-09-28:
Sorry for the noise, I can't reproduce it anymore either.

closed 2024-09-28 by kachurovskiy