FreeCAD/FreeCAD #16888: Too many segfaults in QuickMeasure (here's one more)

AIRCAP, 2024-09-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Everything the user clicks on in the FreeCAD GUI - whether in the part tree, the 3d view, or in selection dialogues in widgets (for example Results of Part_CheckGeometry - but also 3rd party mods) usually makes a selection.

Every selection is passed through QuickMeasure - it gets called from more or less every single click, which exposes this module to a verly large exposure-manifold w.r.t. the stuff it can be called with and the combinations.

For this to be feasible, QuickMeasure has to be coded in a very robust way to deal with this large variety of unknown cases, also in a future proof manner.

This is not the case. As a result, the activation of QuickMeasure in 1.0 has triggered a multitude of regressions, typically ending in a segfault, such as
#14076 
#16794
#16844 
#16820
etc ...
some of which have been fixed while others are hard to reproduce.

This report adds one more. The way Part_CheckGeometry exposes problematic surfaces they are reported not in their immediate partent but in the part they reside in. Clicking on those crashes, since QuickMeasure can only handle selections of things within their immediate parent:

**Steps to reproduce:**
1. Make a new document with at least one Part called "Part"
2. In Python console, enter `Gui.Selection.addSelection(App.ActiveDocument.Name,'Part','SegfaultPlease')`

**Expected Outcome:**
Throw an exception of some kind

**Actual Outcome:**
Segfault from Null pointer dereferencing in QuickMeasure.cpp

The issue is here:

QuickSelection.cpp 122++
```
auto shouldSkip = [](App::DocumentObject* obj) {
        std::string vpType = obj->getViewProviderName();
        auto* vp = Gui::Application::Instance->getViewProvider(obj);
        return (vpType == "SketcherGui::ViewProviderSketch" && vp->isEditing())
            || vpType.find("Gui::ViewProviderOrigin") != std::string::npos
            || vpType.find("Gui::ViewProviderPart") != std::string::npos
            || vpType.find("SpreadsheetGui") != std::string::npos
            || vpType.find("TechDrawGui") != std::string::npos;
    };

```

Aside from the obvious - causing a segfault of obj==NULL  - this is a list of "known bad cases" where quickmeasurement should stay away from- which is a good start but essentially the wrong thinking.  Any future modification and any Mod may add new ViewProviders - any number of which may add shapes that QuickMeasurement can't handle and may crash.

This is just the start. The whole module seems to be coded with blind trust in that the selected objects are all "nails" that can be hammered in - no check if they are pointy or might explode when hammered - as apparent from the previously fixed - and yet unfixed issues above.

I am suggesting treating this issue as a blocker until QuickMeasure has been hardened or deactivated (or deactivated until it has been hardened)

### Full version info

```shell
OS: Ubuntu 20.04.6 LTS (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38844 (Git)
Build type: Debug
Branch: main
Hash: 6e2cd4e73303f817e4bf5e908abe39030b7c5fb3
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
OS: Ubuntu 20.04.6 LTS (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38844 (Git)
Build type: Debug
Branch: main
Hash: 6e2cd4e73303f817e4bf5e908abe39030b7c5fb3
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Curves 0.6.48
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

bgbsww, 2024-09-28:
I agree with the blocker assessment here.  This is a nest of segfaults.

mwganson, 2024-09-28:
Related PR:
https://github.com/FreeCAD/FreeCAD/pull/16855

FEA-eng, 2024-09-28:
#14076 doesn't seem to be related to QuickMeasure, it's about Std Measure, right? But regarding QuickMeasure, can't it just be activated only when selecting vertices, edges, faces or their combinations and nothing else?

AIRCAP, 2024-09-28:
> #14076 doesn't seem to be related to QuickMeasure, it's about Std Measure, right? But regarding QuickMeasure, can't it just be activated only when selecting vertices, edges, faces or their combinations and nothing else?

double checked, yeah #14076 is something else, i might have copy pasted the wrong issue id

hyarion, 2024-09-29:
Here's another one:

https://github.com/user-attachments/assets/ca2ec746-26d8-4133-ab55-faa47dc5e62d

[test_crash.FCStd.zip](https://github.com/user-attachments/files/17177018/test_crash.FCStd.zip)

1. load file
2. select sketch
3. click revolute
4. select z-axis

crash

<details><summary>stack trace</summary>
<p>
```
thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
  * frame #0: 0x000000018a396a60 libsystem_kernel.dylib`__pthread_kill + 8
    frame #1: 0x000000018a3cec20 libsystem_pthread.dylib`pthread_kill + 288
    frame #2: 0x000000018a2dba30 libsystem_c.dylib`abort + 180
    frame #3: 0x000000018a385d08 libc++abi.dylib`abort_message + 132
    frame #4: 0x000000018a375fc0 libc++abi.dylib`demangling_terminate_handler() + 348
    frame #5: 0x000000018a0141e0 libobjc.A.dylib`_objc_terminate() + 160
    frame #6: 0x000000018a3850cc libc++abi.dylib`std::__terminate(void (*)()) + 16
    frame #7: 0x000000018a388348 libc++abi.dylib`__cxxabiv1::failed_throw(__cxxabiv1::__cxa_exception*) + 88
    frame #8: 0x000000018a38828c libc++abi.dylib`__cxa_throw + 308
    frame #9: 0x000000012efb24b0 Measure.so`TopoDS::Edge(S=0x000000016fdfb770) at TopoDS.lxx:71:3
    frame #10: 0x000000012efb3770 Measure.so`Measure::Measurement::length(this=0x0000000166e2c860) const at Measurement.cpp:361:43
    frame #11: 0x0000000167fc4cec MeasureGui.so`MeasureGui::QuickMeasure::printResult(this=0x0000000166e2d2b0) at QuickMeasure.cpp:208:42
    frame #12: 0x0000000167fc412c MeasureGui.so`MeasureGui::QuickMeasure::tryMeasureSelection(this=0x0000000166e2d2b0) at QuickMeasure.cpp:101:5
    frame #13: 0x0000000167fc3d7c MeasureGui.so`MeasureGui::QuickMeasure::processSelection(this=0x0000000166e2d2b0) at QuickMeasure.cpp:81:13
    frame #14: 0x0000000167fc6cd0 MeasureGui.so`QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (MeasureGui::QuickMeasure::*)()>::call(f=(MeasureGui.so`MeasureGui::QuickMeasure::processSelection() at QuickMeasure.cpp:77), o=0x0000000166e2d2b0, arg=0x000000016fdfbe48) at qobjectdefs_impl.h:152:13
    frame #15: 0x0000000167fc6c00 MeasureGui.so`void QtPrivate::FunctionPointer<void (MeasureGui::QuickMeasure::*)()>::call<QtPrivate::List<>, void>(f=(MeasureGui.so`MeasureGui::QuickMeasure::processSelection() at QuickMeasure.cpp:77), o=0x0000000166e2d2b0, arg=0x000000016fdfbe48) at qobjectdefs_impl.h:185:13
    frame #16: 0x0000000167fc6acc MeasureGui.so`QtPrivate::QSlotObject<void (MeasureGui::QuickMeasure::*)(), QtPrivate::List<>, void>::impl(which=1, this_=0x0000000128418470, r=0x0000000166e2d2b0, a=0x000000016fdfbe48, ret=0x0000000000000000) at qobjectdefs_impl.h:418:17
    frame #17: 0x0000000109c15dfc libQt5Core.5.15.8.dylib`___lldb_unnamed_symbol16968 + 820
    frame #18: 0x0000000109c1c96c libQt5Core.5.15.8.dylib`QTimer::timeout(QTimer::QPrivateSignal) + 52
    frame #19: 0x0000000109c11bf8 libQt5Core.5.15.8.dylib`QObject::event(QEvent*) + 104
```
</p>
</details>

luzpaz, 2024-09-29:
cc @hlorus

maxwxyz, 2024-09-29:
@PaddleStroke FYI

PaddleStroke, 2024-09-29:
It seems quick measure needs to be a opt-in rather than opt-out which should solve most issues. I'll check that this week.

AIRCAP, 2024-09-29:
as noted in #16905 - QuickMeasure or its immediate caller also needs a general exception handler to prevent unhandled exceptions from reaching Qt code.

PaddleStroke, 2024-09-30:
- https://github.com/FreeCAD/FreeCAD/issues/14076 is not related
- https://github.com/FreeCAD/FreeCAD/issues/16744 is not related
- https://github.com/FreeCAD/FreeCAD/issues/16794 is fixed
- https://github.com/FreeCAD/FreeCAD/issues/16820 is fixed

Please keep bug reports without clutter since it does not help fixing.

You speak about Part_CheckGeometry  but I your reproduction steps do not mention that. So I'm not sure this mentioned bug is going to be fixed.

The bug in the video above by [hyarion](https://github.com/hyarion) is fixed as well by https://github.com/FreeCAD/FreeCAD/pull/16921.

AIRCAP, 2024-09-30:
a suggestion to future proof this - 

in
https://github.com/FreeCAD/FreeCAD/blob/680b4d60f02f51447e37db7ef2f4323565d7871f/src/Mod/Measure/Gui/QuickMeasure.cpp#L121

would it make sense to add a property "SupportsQuickMeasure" - to selected ViewProviders - and check for its existence?  that way extensions can make sure they play nice with QuickMeasure and want that feature activated.

maxwxyz, 2024-10-06:
@AIRCAP reopened for the opt-in solution as discussed in today's developer meeting

AIRCAP, 2024-10-06:
Note: This has been reopened after discussion QuickMeasure in the developer jits meeting on Sunday 2024-10-06 

The idea is to reduce the QuickMeasrure functionality to when the user is not using any tool.

Background: When a tool is used, it often changes the way the 3d interface works, instructing the user to select or click on things - sometimes things that only exist in the context of a specific workbench. Nevertheless some of them are parts, not covered by #16921 

examples that caused crash in the past was
- selecting an axis in Part Revolve to revolve around as instructed by Part_Revolve
- selecting a broken face out of the report from Part_CheckGeometry
- the FEM workbench FEM_MaterialSolid tool or any Constraint selection, which tell the user to select faces, wires or other topology by clicking on them,
- ...

the risk is that there might be many more actions that are required for the respective workflow of workbenches that might trigger yet untested combinations which could render those workbenches/tools unusable which would be very frustrating to users past release.

the suggestion was to somehow have QuickMeasure not be called or immediately return if any such tool is active, which indicates a deviation from normal workflow - in these cases 
a) the likelyhood the user actually wants to measure or needs measurement is relatively low
b) the likelyhood that something unexpected and untested happens is relatively high
c) if something happens it's not an action the user could easily work around or just not do to avoid a crash/exception

the question is - whats the best function to find out if a tool - any tool - is open (Task dialogue) ?

AIRCAP, 2024-10-06:
the above PR should address this - hopefully

PR #17060 (2024-10-07): Changed QuickMeasurement to be conservative
fix #16888

QuickMeasurement should not measure while tool dialogs are open this includes but is not limited to editing sketches also changed several other sanity checks to be opt-in vs opt-out, as discussed.