FreeCAD/FreeCAD #16890: UX: Confirmation Messages for heavy-duty

gianlucafiore, 2024-09-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

alerts with Confirmation Messages before running tasks that will take too long to run, or take up too much memory space
so that it does not seem that the program crashed. Thinking of end users with little knowledge of the memory management of a software.
Perhaps you could also think about having a progress bar that shows the progress of the execution of a task that involves a stationary time.

In my point of view, this would generate a more pleasant experience that would be taken away by any user, with zero knowledge of software development, who comes to FreeCAD looking for a good alternative to proprietary software

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Spanish/Argentina (es_AR)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2024-09-29:
which operations should have this?

gianlucafiore, 2024-09-29:
it occurs to me when trying to break down an imported DXF, for example, when making a shape_2d_view of a cutting plane that cuts some super complex object, or when trying to turn a mesh with many polygons into a block of parts, also so that if the command is touched by accident, it can be canceled, or suggests that it will take a long time

maxwxyz, 2024-09-29:
Closing this in favor of #5566

closed 2024-09-29 by maxwxyz