FreeCAD/FreeCAD #16895: TechDraw: TechDraw_AreaDimension with 3D reference and "Measure Type" Projected has wrong value

Roy-043, 2024-09-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Open the attached [file](https://github.com/user-attachments/files/17175713/td-area.FCStd.zip) (remove .zip).
2. The area of 21.86mm2 is 1/4th of the unprojected area. This cannot be correct.
3. Change "Measure Type" to "True" to verify.

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38843 (Git)
Build type: Release
Branch: main
Hash: d519f24df9931e4a98d19ee39cdcd6124ae6a7bd
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-09-29:
@WandererFan FYI

WandererFan, 2024-09-29:
Better to send this to the author - @PaddleStroke

WandererFan, 2024-09-30:
AFAIK, the smart dimension tool does not support any 3d references - #14238

closed 2024-11-25 by yorikvanhavre