FreeCAD/FreeCAD #16896: [1.0 RC2] Glass tree view click through?

decryptedchaos, 2024-09-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

My freecad 1.0 RC2 came preinstalled with glass I'm unsure if this is default but thats how it came for me i much prefer the UI.   however there is a problem with glass.   you can't click through it.  

i brought this up in discord but nobody acknowledged it..   When using the mouse to pan or rotate the view.  if your cursor is in the blank area below the tree it won't respond to clicks meaning you can't pan or rotate if your happen to be in that area..   This is going to be a catch point for most users its not intuitive.   it doesn't give any cue why its not responding to imput,  but knowing how it works i was able to notice it happening..  

### Full version info

```shell
OS: EndeavourOS (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: makepkg
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.12.6, Qt 6.7.2, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * Glass
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FreeWilly2022, 2024-09-28:
I am no dev, but I know its picking up the glass mod from your previous installation. I just checked mine and it has the mods from my previous versions, but of course no glass because I have never used that mod.

decryptedchaos, 2024-09-28:
> I am no dev, but I know its picking up the glass mod from your previous installation. I just checked mine and it has the mods from my previous versions, but of course no glass because I have never used that mod.

that's just it its a fresh install.   never had any other version installed on this particular install..   it could be something they are packing with it in the arch repo idk

luzpaz, 2024-09-29:
I would not use glass, as it is not maintained. 

@chennes maybe it's time to deprecate it ?

adrianinsaval, 2024-09-29:
Glass does not come preinstalled with freecad, neither our distribution nor arch's package I don't know how you got it there, you must have installed it at some point in the past. Keep in mind that even if you uninstalled freecad a long time ago old addons would still persist as they are stored in your home directory and are not cleaned up by the package manager.

> But as a test i disabled it, and with it disabled i have no tree at all..

you may reset your preferences or right click on an empty toolbar area and click the checkbox to enable the tree again. Then you can click the "toggle overlay" button in your tree view panel or almost any other panel for that matter to get glass like behavior but even better since click through works with it if you enable transparent mode once you are in overlay mode
[image]
[image]

In any case, this is not a bug so I'll close it.

btw I agree with marking glass as deprecated once we release 1.0, while it's not technically going to stop working it serves no purpose anymore.

closed 2024-09-29 by adrianinsaval

luzpaz, 2024-09-29:
We could have the Addon Manager Auto-disable it or give the user the option to proceed at their own risk

adrianinsaval, 2024-09-29:
I would suggest creating a new issue about a deprecation mechanism for addons

luzpaz, 2024-09-29:
* https://github.com/FreeCAD/FreeCAD-addons/issues/276

adrianinsaval, 2024-10-05:
> * [Prune unmaintained addons FreeCAD-addons#276](https://github.com/FreeCAD/FreeCAD-addons/issues/276)

I believe there is a process to mark addons as deprecated but no mechanism to notify the user about installed addons that are not compatible