FreeCAD/FreeCAD #16898: Section Issues

pcharliep01, 2024-09-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When I try to cross section a drawing I get a series of circles appear that I can't explain, can't remove and get no error massages for.

### Full version info

```shell
When I try to cross section a drawing I get a series of circles appear that I can't explain, can't remove and get no error massages for. As far as I know there are no issues with the model. I insert the model into TechDraw, orientate it for a top view, then try to cross section this view to the left whish then fails. So far I have only had this happen to this model so I assume that there is something wrong with this model but can't find anything wrong. I can send a copy of the .FCStd file if required.
```

### Subproject(s) affected?

TechDraw

### Anything else?

[image]
[image]
[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-09-29:
Please provide the example file as .zip (just upload it here in a comment) and also the full about info of the FreeCAD version you are using.

pcharliep01, 2024-09-29:
The version is 1.0.0RC1. And no it isn't working properly on ver 0.21.2 when I try the same example on that version I get a line going out on an angle but no circles.

Is it working as expected in the stable 0.21.2 version?

luzpaz, 2024-09-30:
@pcharliep01  
* Click on this https://github.com/FreeCAD/FreeCAD/issues/16898  
* Zip up your example in a .zip file
* Drag/drop .zip file into the github textbox to auto-attach it to this ticket
* Edit the original post and add your [Full About](https://wiki.freecad.org/About) info in the section named "Full version info"

WandererFan, 2024-09-30:
There was a bug a while back where bspline arcs were being drawn as full circles.  I don't know if the fixes are in rc1 or not.

The example file would be really helpful.  Just add ".zip" to the end of the file name and github should accept it.

pcharliep01, 2024-09-30:
[Example Ver 1-0RC.zip](https://github.com/user-attachments/files/17196496/Example.Ver.1-0RC.zip)
this a a 7zip file I had to change the extension

luzpaz, 2024-09-30:
@pcharliep01 Thanks. One last thing:  

* Edit the original post and add your [Full About](https://wiki.freecad.org/About) info in the section named "Full version info"

pcharliep01, 2024-09-30:
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/Australia (en_AU)
Installed mods: 
  * Curves 0.6.44
  * pcb 6.2023.1
  * sheetmetal 0.4.24

pcharliep01, 2024-09-30:
Hi, I just updated the GitHub with a zip of the project and the details of the FreeCad I am using.
Thanks

WandererFan, 2024-10-01:
Lots of geometry errors in the body.  That is going to make getting a good result difficult.

[image]

pcharliep01, 2024-10-02:
Hi, the example model is from an exercise from a YouTube channel I subscribe to. He did two of the same model, one in Ver0.21 and the other in Ver1.0RC to compare the differences,  I followed his way of modeling both and got the same result on both. As far as I can tell I came across no errors in the final models. I then tried to draw them in Techdraw and this is when I came across the problem. I drew the Model created in ver 0.21 using the techdraw in ver0.21 and had no problems but when I tried using the Model from Ver1.0 in Techdraw 1.0 it failed and when a tried using the model created in ver0.21 in the Ver1.0 Techdraw it also failed !!
I appreciate your assistance, Thanks

Example.Ver.1-0RC_CheckGeom.png (view on web)<https://github.com/user-attachments/assets/e50b4640-d0d5-42b0-bc7c-eaf705718c97>

WandererFan, 2024-10-03:
Does your v0.21 use the same version of OCC as your v1.0?

pcharliep01, 2024-10-03:
Sorry I don't know what that means but if you let me know where to find the info I will let you know.

luzpaz, 2024-10-04:
@pcharliep01 it simply means that we're trying to ascertain if the 3rd party dependency OCC (OpenCasCade which is a geometry kernel) is the same version between the 2 FreeCAD versions you used. You can see this in the About info you posted above for 1.0RC1 version:
>  Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, **OCC 7.7.2**

We need you to paste the [About](https://wiki.freecad.org/About) info of the v0.21.x that you used. Can you please do that?

pcharliep01, 2024-10-04:
Here is the info you requested.
Ver 0.21.2

  *
Boost 1_67
  *
Coin3D 4.0.1
  *
Eigen 3.3.90
  *
FreeType 2.10.1
  *
KDL
  *
libarea
  *
Open CASCADE Technology 7.6.3
  *
Point Cloud Library
  *
PyCXX 7.1.7
  *
Python 3.8.10
  *
Qt for Python (PySide) 5.15.0
  *
Qt 5.15.2
  *
Salome SMESH 7.7.1.0
  *
Qt for Python (Shiboken) 5.15.0
  *
vtk 8.2.0
  *
Xerces-C 3.2.2
  *
Zipios++
  *
zlib 1.2.11

Ver 1.0.0RC1

  *
Boost 1_82
  *
Coin3D 4.0.3
  *
Eigen 3.4.0
  *
FreeType 2.12.1
  *
KDL
  *
libarea
  *
Open CASCADE Technology 7.7.2
  *
Point Cloud Library 1.14.0
  *
PyCXX 7.1.9
  *
Python 3.11.9
  *
Qt for Python (PySide) 5.15.8
  *
Qt 5.15.13
  *
Salome SMESH 9.8.0.2
  *
Qt for Python (Shiboken) 5.15.8
  *
vtk 9.2.6
  *
Xerces-C 3.2.5
  *
Zipios++
  *
zlib 1.2.13

luzpaz, 2024-10-04:
> Does your v0.21 use the same version of OCC as your v1.0?

@WandererFan, different OCC versions 7.6.3 vs 7.7.2 (respectively)

WandererFan, 2024-10-07:
I am ~90% certain this is an upstream issue in OpenCascade.  If we use the perspective projector with a long focus, the problem goes away.  Same with changing the direction slightly or moving the the section origin by 0.001 mm.

[image]

I trapped the output of the hlr and it has a giant unwanted edge.  The edge says that it is an almost complete circle with a length of > 1 m which seems silly.

[image]

WandererFan, 2024-10-07:
I have the troublesome shapes saved.  I'll try to make a DRAWEXE script so we can submit a bug to OCC.

FWIW, I'm using OCC 7.8.0 to debug.

OS: Linux Mint 21.2 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38903 +3 (Git)
Build type: debug
Branch: fixBadCircles16980
Hash: ed2eb34930141d83fd67c032a7b43b36924d7fc5
Python 3.10.12, Qt 5.15.2, Coin 4.0.0, Vtk , OCC 7.8.0
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * fasteners 0.5.21

WandererFan, 2024-10-10:
If somebody could give PR #17182 a test, that would be great.