FreeCAD/FreeCAD #16904: Part Design extrusion, "Wire is not closed", small bug that can be resolved.

bigi111, 2024-09-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In Part Design, I created two B-spline curves in a sketch. Then I joined the two end with two vertical lines. When I try to extrude the sketch, I have the message : "Wire is not closed". If I enter the sketch and just move one vertical line, quit the sketch, then the extrude works.
I wanted to add two simple Freecad files to show the problem.
You can download them from here :
https://kdrive.infomaniak.com/app/share/503070/afb1ec80-7a9c-4d00-abe5-58f808ff66cb
https://kdrive.infomaniak.com/app/share/503070/87bc9c7d-d2ac-424a-bc3a-7c642fba4dae
I saw only after writing the links above the I can transfer the FreeCAD files in a .zip file, which I do below.

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: French/Switzerland (fr_CH)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * symbols_library
  * ExplodedAssembly
  * Curves 0.6.5
  * fasteners 0.4.56
  * MeshRemodel 1.8919.0
  * Lithophane
```

### Subproject(s) affected?

PartDesign

### Anything else?

[Part_Design_small_bug.zip](https://github.com/user-attachments/files/17178747/Part_Design_small_bug.zip)
One file is 53'440 byte long.
The second one is a simplification and is 18'099 bytes long.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-09-29:
This has a lot of similarities to https://github.com/FreeCAD/FreeCAD/issues/16127 where you'll see my investigation shows this bug has been in existence for a long time (since at least 0.19 dev cycle).

pixelwaster, 2024-09-29:
Can confirm it does as @bigi111 describes.  (1.0RC2 Windows 10 full version at end)
Also, I moved one point of the vertical line (.01 .01 0.0) mm and the wire was closed and padded. I zoomed really far in.
I also only added a length constraint to the vertical line Part_Design_extrusion_small_bugC.Body001.Sketch001.Edge135 (closest to Y axis) and the wire was closed and padded

Not sure if this will help but found by using ^C instead of Right Click->Copy. In geoList.append one of the App.Vector values change after adding constraint. 

Before any changes
```
# Copied from sketcher. From:
#objectStr = App.getDocument('Part_Design_extrusion_small_bugC').getObject('Sketch001')
lastGeoId = len(ActiveSketch.Geometry)

geoList = []
geoList.append(Part.LineSegment(App.Vector(2.224062, 25.558462, 0.000000),App.Vector(2.224062, 22.558447, 0.000000)))
objectStr.addGeometry(geoList,False)
del geoList

objectStr.addConstraint(Sketcher.Constraint('Vertical', lastGeoId + 0))
```
Added length constraint
```
# Copied from sketcher. From:
#objectStr = App.getDocument('Part_Design_extrusion_small_bugC').getObject('Sketch001')
lastGeoId = len(ActiveSketch.Geometry)

geoList = []
geoList.append(Part.LineSegment(App.Vector(2.224062, 25.558462, 0.000000),App.Vector(2.224062, 22.558462, 0.000000)))
objectStr.addGeometry(geoList,False)
del geoList

constraintList = []
constraintList.append(Sketcher.Constraint('Vertical', lastGeoId + 0))
constraintList.append(Sketcher.Constraint('DistanceY', lastGeoId + 0, 2, lastGeoId + 0, 1, 3.000000))
objectStr.addConstraint(constraintList)
del constraintList

```

OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * AddonManager
  * Assembly
  * BIM
  * btl 0.9.9
  * CAM
  * CurvedShapes 1.0.10
  * Curves 0.6.47
  * Draft
  * FeedsAndSpeeds 0.5.0
  * Fem
  * Help 1.0.3
  * Idf
  * Import
  * Inspection
  * lattice2 1.0.0
  * Manipulator 1.5.7
  * Material
  * Measure
  * Mesh
  * MeshPart
  * OpenSCAD
  * Part
  * PartDesign
  * Plot 2022.4.17
  * Points
  * ReverseEngineering
  * Robot
  * sheetmetal 0.4.26
  * Show
  * Silk 0.1.5
  * Sketcher
  * Spreadsheet
  * Start
  * Surface
  * TechDraw
  * Test
  * Tux
  * Web

maxwxyz, 2024-10-01:
@AjinkyaDahale FYI

bigi111, 2024-10-02:
I did some more tries and can reproduce the bug in many situations.
1) Document, new
2) New body, new sketch (XY plan, but other plans give the same result)
3) I draw a B-spline by knots (being a curve, like a function without over crossing itself)
4) I draw a similar B-spline under the first one.
5) I draw a line (G, L) from the first point of the first B-spline to the first point of the second B-spline
6) I draw a line (G, L) from the last point of the first B-spline to the last point of the second B-spline
It is important that the two lines do not become vertical or get any other constrains.
7) I close the sketcher construction
8) When I try to do a Pad (an extrusion) and click on OK, I get  : ”Wire is not closed”.

If instead of “B-spline by knots” I do “Create arc by center”, I get the same result.
If I use “B-spline by control points”, I couldn’t reproduce this bug.
I tried also with one “B-spline by knots” and one “B-spline by control points” without reproducing the bug.
With one “B-spline by knots” and one “Create arc by center”, connected by lines at their end points, I reproduced the bug.
Each time I verified that the curve was well closed in two ways. In the sketcher, in the constraints panel and by controlling that after moving one line I could pad the sketch.
If I should to more testing, tell me.

AjinkyaDahale, 2024-10-06:
> I did some more tries and can reproduce the bug in many situations.
> 

Cannot reproduce with

```
OS: Ubuntu Core 22 (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38898 (Git) Snap 1157
Build type: Release
Branch: main
Hash: 32e09d9554db26414697ed19e0189ccab4ea9029
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

> If I should to more testing, tell me.

@bigi111 can you attach a file just before the pad that fails?

bigi111, 2024-10-07:
I tested with the same version as above :
OS: Ubuntu 24.04.1 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38898 (Git) AppImage
Build type: Release
Branch: main
Hash: 32e09d9554db26414697ed19e0189ccab4ea9029
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: French/Switzerland (fr_CH)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

The "just-before-failing-pad.zip" fails also on my computer.
If I edit the sketch, move any point, the pad works.
If I end pressing "Cancel" to not get the pad, then press two times on Ctrl+Z to undo the point move, then again the pad doesn't work and give the same error "wired is not closed".
So this bug can well be reproduced.
Some update must be missing when quitting the sketch.

If I don't move any point, but add a length constraint, the the pad works.
Undoing by pressing two Ctrl+Z, then the bug is again here and the pad doesn't work.